platformio/platformio-atom-ide #100: Popups/Dialogs not working

snej0l, 2016-04-21:
I use PlatformIO on my Mac and have the following problem: When open a dialog / window I can not choose anything there and not click on any button. This relates both to the window for the Serial Monitor and the Donate window. I can edit text in the background text editor but the dialog is always in the foreground.

ivankravets, 2016-04-21:
1. What is the version of Atom?
2. Have you installed our IDE bundle with Atom or used http://atom.io?

snej0l, 2016-04-21:
> What is the version of Atom?
> Have you installed our IDE bundle with Atom or used http://atom.io?

The current IDE bundle

ivankravets, 2016-04-21:
Please try Atom from http://atom.io

snej0l, 2016-04-22:
I found the problem. The theme in Atom was the problem. When I switched to the default theme everything works fine.

Thanks for helping

ivankravets, 2016-04-29:
With which theme have you had a problem?

snej0l, 2016-04-30:
https://atom.io/themes/native-ui

ivankravets, 2016-06-23:
It looks that it can be fixed in `native-ui` theme. I've checked and it works fine.

closed 2016-06-23 by ivankravets