platformio/platformio-atom-ide #104: Rename debian package, enable installation side by side with atom

arthurlutz, 2016-04-25:
I already have atom and was surprised to have dpkg or ubuntu software manager offer an upgrade when downloading platformio. It would be nice for the two apps to be different and use them independently. If this is not desired, then it should be more explicit in the download page, for example "Download a customized atom package that will auto-install platform.io extensions". And then offer some "manual" way of adding platform.io to an existing atom install.

ivankravets, 2016-04-25:
We have this info on the [Get Started](http://platformio.org/get-started) page.

> PlatformIO IDE is built on top of GitHub's Atom "hackable" text editor. If you have Atom already installed please use [PlatformIO IDE package for Atom](http://docs.platformio.org/en/latest/ide/atom.html#installation).

ivankravets, 2016-06-23:
You can install Atom from the own system package manager. Please follow to "Settings > Install >  Packages" and type "platformio-ide".

closed 2016-06-23 by ivankravets