platformio/platformio-atom-ide #105: Cannot install

positron96, 2016-04-25:
First launch of installer said that I do not have Python installed (which was true). After I installed Python and rebooted launching installer opens atom window with blue hint "PlatformIO IDE installation suspended. Seems like PlatformIO IDE is already being installed in another window". How do I fix it?

Logs in  SquirrelSetup.log say ```SingleGlobalInstance: Failed to grab lockfile, will retry: C:\Users\positron\AppData\Local\Temp\.squirrel-lock-3066DD322F5834BF3DC2D39E6F983A2E5C4919D7: System.IO.IOException: The process cannot access the file 'C:\Users\positron\AppData\Local\Temp\.squirrel-lock-3066DD322F5834BF3DC2D39E6F983A2E5C4919D7' because it is being used by another process.```
but there are no open handles to this file after I close atom window.

OS: Windows 7 x64

ivankravets, 2016-04-25:
Could you try to close Atom and  remove `C:\Users\positron\.atom`?

positron96, 2016-04-26:
Seems like it helped, thank you. But the requirement of python should probably be more clearly visible before installation (or a separate full installation bundle with python would be preferrable in windows).

ivankravets, 2016-04-26:
> the requirement of python should probably be more clearly visible before installation

We have it http://docs.platformio.org/en/latest/ide/atom.html#installation

> or a separate full installation bundle with python would be preferrable in windows

Thanks, we are working on it.

ivankravets, 2016-04-26:
@orgkhnargh please reproduce this issue:

1. Test in Windows VM
2. Delete Python
3. Run PlatformIO IDE
4. Try to Install Python
5. Check that lockfile has been deleted.

closed 2016-04-29 by ivankravets