platformio/platformio-atom-ide #106: Keyboard stuck after use serial monitor

saspol, 2016-04-26:
Im trying to debug esp8266 board with serial monitor.
After few times open monitor i got this error and then my macbook keyboard dont respond any key.
Keyboard completely stuck and only reboot can help.
WIth arduino boards serial monitor works fine.
I use OSX 10.8.5

Exception in thread rx:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/Cellar/platformio/2.8.6/libexec/vendor/lib/python2.7/site-packages/serial/tools/miniterm.py", line 411, in reader
    self.console.write(text)
  File "/usr/local/Cellar/platformio/2.8.6/libexec/vendor/lib/python2.7/site-packages/serial/tools/miniterm.py", line 60, in write
    self.output.write(s)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2400' in position 0: ordinal not in range(128)

ivankravets, 2016-04-26:
We added `raw` option to serial monitor https://github.com/platformio/platformio-atom-ide/commit/943c3755d340d3fdfcee6e96e2338a00a4517ae9

PlatformIO IDE 1.2.0 will be released this week.

ivankravets, 2016-04-28:
PlatformIO IDE is going to be released today!

closed 2016-04-28 by ivankravets