platformio/platformio-atom-ide #108: Uncaught Error: Failed to get boards (OS X)

arkzor, 2016-04-29:
Hi !

Just tried to install platformio ide (http://platformio.org/get-started/ide?dl=platformio-atom-mac.zip) and I get this error when I try to begin a new project. I already read the other thread about this error on Windows 10 but the solution does not seem to work on OS X... When I try to uninstall the package from Atom (like the other user did) I get this error : Failed to delete platformio-ide: Does not exist. But it is listed in the installed packages...

The platformio command line is working fine from the terminal window.

I also tried to download a clean Atom at first and then install platformio-ide and platformio-ide-terminal but I still get the same error.

Thank you for your help.

Nicholas

[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.7.1
**System**: Mac OS X 10.11.4
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.2.0

### Stack Trace

Uncaught Error: Failed to get boards

```
At /Users/nicko/.atom/packages/platformio-ide/lib/utils.js:173

Error: Failed to get boards
    at getBoards (/Users/nicko/.atom/packages/platformio-ide/lib/utils.js:173:13)
    at InitializeNewProjectView.initializeBoardsSelect (/Users/nicko/.atom/packages/platformio-ide/lib/init/view.js:67:20)
    at new InitializeNewProjectView (/Users/nicko/.atom/packages/platformio-ide/lib/init/view.js:63:10)
    at command (/Users/nicko/.atom/packages/platformio-ide/lib/init/command.js:34:14)
    at atom-workspace.platformioIdeInitializeNewProject (/Users/nicko/.atom/packages/platformio-ide/lib/main.js:71:54)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/Applications/Atom.app/Contents/Resources/app.asar/src/command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (/Applications/Atom.app/Contents/Resources/app.asar/src/command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/Applications/Atom.app/Contents/Resources/app.asar/src/command-registry.js:160:19)
    at runAtomCommand (/Users/nicko/.atom/packages/platformio-ide/lib/utils.js:109:24)
    at HTMLAnchorElement.callback (/Users/nicko/.atom/packages/platformio-ide/lib/home-screen/quick-links.js:36:25)
```

### Commands

```
     -0:00.5.0 platformio-ide:initialize-new-project (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
```

### Config

```json
{}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.9.3
build, v0.60.0
language-ini, v1.16.0
linter, v1.11.4
linter-gcc, v0.6.14
platformio-ide, v1.2.0
platformio-ide-terminal, v2.0.9
tool-bar, v0.4.0

# Dev
No dev packages
```

orgkhnargh, 2016-04-29:
> When I try to uninstall the package from Atom (like the other user did) I get this error : Failed to delete platformio-ide: Does not exist. But it is listed in the installed packages...

How exactly do you uninstall the package? With `apm uninstall` or from Atom Settings GUI?

arkzor, 2016-04-29:
Hi !

With atom settings, with apm I succeeded but after reinstallation same
problem.

Nicholas

ivankravets, 2016-04-29:
@arkzor could you remove `~/.atom` folder and restart IDE?

arkzor, 2016-04-29:
@ivankravets : Well, than you very much ! :) It now works !

closed 2016-04-29 by arkzor

vannut, 2016-08-24:
Am getting the same error. But removing the ~/.atom folder and restarting does not the trick. 
It resets the complete atom install and the need to reinstall the platform.io package.

Any other possible ways to solve this?

ivankravets, 2016-08-24:
@vannut what is your version of PlatformIO CLI?

vannut, 2016-08-24:
`PlatformIO, version 2.10.3`
And in atom: platformio-ide: 1.4.0

ivankravets, 2016-08-24:
Do you use custom PlatformIO CLI? I see that the version is different than the latest.

vannut, 2016-08-24:
I first only installed the package through atom.io. And afterwards the CLI through homebrew.

ivankravets, 2016-08-24:
You don't need to install CLI via Brew. It's built-in in PlatformIO IDE.

Please `brew uninstall platformio` and restart IDE

vannut, 2016-08-24:
Uninstalled platformio via homebrew. 
Still getting the error: https://gist.github.com/vannut/7a9918a00abab2e668bb552413e5c422

ivankravets, 2016-08-24:
Please open PlatformIO IDE Terminal. What do you see with this command?
```
pio boards
```

vannut, 2016-08-24:
Nope same error: https://gist.github.com/vannut/284ad5f3485c8f208815a3210647114d

ivankravets, 2016-08-24:
You have broken board manifest in `~/.platformio/boards/`. I've just added error warning when an user wrote broken JSON manifest. This is implemented in the latest PlatformIO 3.0 development version. See https://github.com/platformio/platformio/commit/aa19b1c424f9a2baa9462269358cd42a7da889d2

Please switch to development version http://docs.platformio.org/en/stable/installation.html#development-version

vannut, 2016-08-24:
That's it!! What a waste of time debugging this:) I like the #aa19b1c commit :)

ivankravets, 2016-08-24:
@vannut Happy Coding with PlatformIO! 😊

vannut, 2016-08-24:
Will do! It's finally working and I already feel the relief of the burden of the arduino ide :) Now only figuring out how a library installed wit pio lib install <int> can be used in an .cpp

ivankravets, 2016-08-24:
> Now only figuring out how a library installed wit pio lib install can be used in an .cpp

Just place `#include <LibHeader.h>` on the top of project file. PlatformIO will find this library automatically, built and link to final firmware.

vannut, 2016-08-24:
Nope :(
[image]

ivankravets, 2016-08-24:
See http://docs.platformio.org/en/stable/ide/atom.html#smart-code-linter

vannut, 2016-08-24:
Ah damn. I looked at the docs about the library manager. Would be a good idea to put that there too :)

ivankravets, 2016-08-24:
@vannut how did you install a library? This is really a bug in PlatformIO IDE. It doesn't monitor new PlatformIO 3.0 library storages.

In any case, I updated the docs https://github.com/platformio/platformio/commit/a57ed939234f6fd12d312d5d6fe879502d4c98f8

vannut, 2016-08-24:
👍  I installed it via the Library manager:`pio lib search RN_2483` and then  `pio lib install 405`

ivankravets, 2016-08-24:
Please show me an output of `pio lib list`

vannut, 2016-08-24:
```
pio lib list
[ ID  ] Name             Compatibility         "Authors": Description
--------------------------------------------------------------------------------------------------------------
[ 405 ] Sodaq_RN2483     arduino, atmelavr, atmelsam, espressif, intel_arc32, microchippic32, nordicnrf51, tee
nsy, timsp430 "Alex Tsamakos, SODAQ, Kees Bakker": An Arduino library for the Microchip RN2483.
[ 416 ] TinyGPS          arduino, atmelavr     "Mikal Hart": A compact NMEA (GPS) parsing library
```

After a rebuilt it now finds both library's.
Would be a nice feature to know what header file you need to include.

ivankravets, 2016-08-24:
Hm... What is your `pio --version`?

> Would be a nice feature to know what header file you need to include.

In this case, please use PlatformIO Library Registry where is explained in examples which headers need to use:
* http://platformio.org/lib/show/405/Sodaq_RN2483
* http://platformio.org/lib/show/416/TinyGPS

vannut, 2016-08-24:
`PlatformIO, version 2.11.2`

ivankravets, 2016-08-24:
Did you install that libraries from PlatformIO IDE Terminal?

vannut, 2016-08-24:
Jups. I clicked on button in the ribbon:
[image]

Then the terminal popped up:
`pio lib search rn2483`
`pio lib install 405`

ivankravets, 2016-08-24:
Thanks, that really a bug. I'll test it.

vannut, 2016-08-24:
:) Glad to help :)

DeniseBryson, 2016-11-29:
I get the same error. Uncaught error: Failed to get boards
Tried to delete ~/.atom, restart Atom and reinstall PlatformIO IDE but the error stayed the same.
I use Ubuntu 16.04 LTS.

The version shown on the Welcome screen shows this.
IDE 1.6.0 | CLI Failed to retrieve

`pio boards` puts out a long list of different platforms

`pio --version`
PlatformIO, version 3.1.0

There is some confusion on the atom --version.
Using a normal terminal or just checking under Help. 
It shows: Atom    : 1.12.5
Using the PlatformIO IDE Terminal I get:

> cooper@CooperBook:~$ atom --version
> v6.5.0

ivankravets, 2016-11-30:
@DeniseBryson please report to https://community.platformio.org