platformio/platformio-atom-ide #112: Unable to start the PlatformIO IDE

MichMich, 2016-05-10:
Hi guys,

I'm trying to het PlatformIO running on my MacBook (El Capitan). Unfortunately, I get an error when starting:
[image]

I've tried reinstalling setuptools and platformio, but unfortunately with no luck.

ivankravets, 2016-05-10:
Please try `rm -rf ~/.atom` and restart. Did it help you?

MichMich, 2016-05-10:
Already tried that. Same problem. :(

ivankravets, 2016-05-10:
Very strange problem. Something is wrong with setuptools.

1. Open system OS Terminal (non IDE Terminal) and run `sudo pip install -U setuptools`
2. `rm -rf ~/.atom`
3. Restart

MichMich, 2016-05-10:
```
$ sudo pip install -U setuptools
Requirement already up-to-date: setuptools in /usr/local/lib/python2.7/site-packages

$ rm -rf ~/.atom
```
Still same problem.

Also interesting:
```
$ platformio
-bash: platformio: command not found

$ echo $PATH
/Users/Michael/.composer/vendor/bin:/usr/local/share/npm/lib/node_modules/grunt-cli/bin:/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
```

ivankravets, 2016-05-10:
Could you provide TeamViewer session for debug? Mail me@ikravets.com. Thanks.

MichMich, 2016-05-10:
TeamViewer credentials in your mail.

svogl, 2016-06-23:
I had the same problem on my Capitan with german OS settings; this command line worked for me:

`LC_ALL=C /Applications/Atom.app/Contents/MacOS/Atom`

ivankravets, 2016-06-23:
Try to this line to your Shell profiles (`~/.bash_profile` or `~/.zshrc`):
```
export LC_ALL=en_US.UTF-8
```

MichMich, 2016-06-23:
I agree that this solves the issue, but should't PlatformIO include a check for this?

ivankravets, 2016-06-23:
> but should't PlatformIO include a check for this?

Of course, it should check. That is why this issue marked as `bug`.

ivankravets, 2016-06-23:
@MichMich Could you try one idea?

1. Close Atom
2. Please disable `export LC_ALL=en_US.UTF-8` in `~/.bash_profile`
3. Open Atom. Now, it should raise an error.

------
Next:
1. Close Atom
2. Please edit this file [~/.atom/packages/platformio-ide/origin/lib/main.js](https://github.com/platformio/platformio-atom-ide/blob/develop/lib/main.js#L55)
3. Place this line `process.env.LC_ALL = 'en_US.UTF-8';` below `process.env.PATH = process.env.Path;`
4. Start Atom

====

Does it help?

MichMich, 2016-06-23:
The file `~/.atom/packages/platformio-ide/origin/lib/main.js ` does not exist. (There isn't an `origin` folder.)
I tried that change in `~/.atom/packages/platformio-ide/lib/main.js ` and that indeed solved the issue.

ivankravets, 2016-06-23:
> does not exist. (There isn't an origin folder.)

Sorry, it was my mistake.

> I tried that change in ~/.atom/packages/platformio-ide/lib/main.js and that indeed solved the issue.

Do you mean, that `process.env.LC_ALL = 'en_US.UTF-8';` helped you?

ivankravets, 2016-06-23:
Could you try to restart Mac? It seems that OS caches system environment variables.

MichMich, 2016-06-23:
> Do you mean, that process.env.LC_ALL = 'en_US.UTF-8'; helped you?

Yes, this is correct.

ivankravets, 2016-06-23:
@svogl Could you try to replace this file `~/.atom/packages/platformio-ide/lib/main.js` with https://github.com/platformio/platformio-atom-ide/blob/develop/lib/main.js ?

closed 2016-06-23 by ivankravets

svogl, 2016-06-23:
@ivankravets Work for me as well - I copied the modified platformio-ide package into a clean .atom subdirectory and the base installation has completed successfully.

ivankravets, 2016-06-23:
@svogl Thanks a lot for the report! I'll release new IDE version as soon as possible.

commit 5538350a (2016-06-23): Fix issue with Python and system locale // Resolve #112