platformio/platformio-atom-ide #113: "Seems like PlatformIO IDE is already being installed in another window." on startup

hexaguin, 2016-05-11:
Every time I start Atom, I get this message. Everything seems to work fine, but the message is getting annoying. What is causing it?

ivankravets, 2016-05-11:
What is your OS?

hexaguin, 2016-05-11:
Windows. I actually managed to fix the issue by deleting install.lock in the PlatformIO package directory and letting the installation finish again. Everything seems to work perfectly now.

ivankravets, 2016-05-11:
We will try automatically to delete expired `install.lock`. Thanks.

sansil, 2016-05-25:
Hi, I have the same issue, but i'm using OS X 10.10.3
I can't find install.lock where is located ?.  Thanks in advance

ivankravets, 2016-05-25:
@sansil it is located in `~/.atom/packages/platformio-ide/install.lock`

sansil, 2016-05-25:
thanks! it works :D

ivankravets, 2016-05-25:
Thanks for the report. We will fix this issue in the next release.

closed 2016-06-23 by ivankravets

geekerator, 2017-09-05:
I also get this message, but in another way. I:
  Started atom (with Platformio previously installed) and did an update to the latest core
  Platformio Core flashing download icon never stopped (after about 2 hrs) - and reappears on any restart
  No work can be done while Installing is going on, so I
  Quit and restarted while randomly clicking on atom's menu items, then
  This message comes up. I don't seem to have an install.lock file anywhere
My specs are: openSuSE 42.3 and kernel 4.4.79-19. Any suggestions would be greatly appreciated...

ivankravets, 2017-09-05:
Could you remove `~/.platformio/penv` folder and restart Atom?

commit fb3cea09 (2016-06-23): Cleanup dead installation lock file // Resolve #113