platformio/platformio-atom-ide #115: Initializing nodemcu project

grahen, 2016-05-16:
Hello!

I just bought my self a nodemcu v2 card and thought I would give pio a spin on building things for it. Following a couple of the starter up tutorials on the web won't do it however. 

I get the following compilation error:

*.pioenvs/nodemcuv2/FrameworkArduinoVariant/pins_arduino.h:29:31: fatal error: ../generic/common.h: No such file or directory*

no matter how I search in the libs and platforms for the pio I can't figure out how to automatically get hold of the named common.h file. I did however find it in this repo https://github.com/esp8266/Arduino. Cloning this and copying the generic folder to the nodemcu folder did the trick. 

Even more compilation errors happen if I choose to include a wifi control lib. I simply have run out of ideas where to go next.

I'm on a Mac if that has something todo with it, also updated atom and pio to the latest versions, or so I think. 

Kind regards
Henrik

ivankravets, 2016-05-16:
Could you share whole project here? Including `platformio.ini`. Thanks.

grahen, 2016-05-17:
Hi there, of course! =). 

I pushed to a repo on my account: https://github.com/grahen/nodemcu2

This one have the problem with a missing common.h file, and as stated in the previous post, if I add the *generic* folder from the Arduino esp8266 project it will work. Either I'm doing something really wrong or I just have bad luck =). 

Thanks for helping!

ivankravets, 2016-05-17:
Have you installed PlatformIO CLI before? Please open system terminal and type `platformio --version`.

grahen, 2016-05-17:
Yepp, 
output is:
*PlatformIO, version 2.8.6*

ivankravets, 2016-05-17:
Please uninstall previous version using System Terminal (not IDE) via:
```
[sudo] pip uninstall platformio
```

Try again IDE.

ivankravets, 2016-05-17:
I forgot to mention that PlatformIO IDE contains own the latest version of PlatfromIO CLI. You don't need to install it separately.

grahen, 2016-05-17:
Alright, hmm, seems that I don't have pip installed on my Mac here. 

What about using apm? I.e. apm uninstall platformio?

ivankravets, 2016-05-17:
> Yepp, 
> output is:
> PlatformIO, version 2.8.6

Sorry, I don't understand how did you install PlatformIO CLI before. Do you use the latest PlatformIO IDE version? PlatformIO IDE != PlatformIO CLI.

ivankravets, 2016-05-17:
Try to `rm -rf ~/.atom` and restart IDE.

grahen, 2016-05-17:
Just figured out that I must have installed platformio through homebrew. Performing upgrade now using brew. 

Once it is finished I will try again.

ivankravets, 2016-05-17:
Good, then `brew uninstall platformio`.

ivankravets, 2016-05-17:
You don't need 2 copies of PlatformIO CLI.

grahen, 2016-05-17:
Great success, it worked by cleanly and neatly upgrading everything. 

Now on IDE 1.2.2 and CLI 2.9.1 and things compile as expected :-) 

Many thanks for your patience!

closed 2016-05-17 by grahen

ivankravets, 2016-05-17:
Happy Coding with PlatformIO! 😊