platformio/platformio-atom-ide #117: Clang linter keeps warning about c++11-only feaures

frnhr, 2016-05-19:
I'm new to PlatformIO (and love it), so please forgive me if I'm being n00b here.

I'm using PlatformIO for Arduino, with mostly default settings. There are some C++11 - only features in the code. Arduino compiles with `-std=gnu++11` flag by default, so it compiles successfully.

But the linter still warns about features only available with this flag:
[image]

I have tried to add `-std=gnu++11` flag to `.clang_complete` file manually, no results.

Is there a way to get rid of these warnings, without giving up all other warnings as well (as would be the case with `"gccSuppressWarnings": true`)?

ivankravets, 2016-05-19:
Could you provide the code/platformio.ini to reproduce this issue? Thanks.

frnhr, 2016-05-19:
A clean installation of PlatformIO, on a clean Os X (virtual machine):

 * start a new project (I'm using Arduino Mega 2560)
 * create a new header file `src/something.h`
 * paste this code in:   
    
```
class A
{
public:
  int a_list[5];
  A() :
    a_list{1, 2, 3, 4, 5}  // This line
  {};
};
```
 * save

> **GCC warning** extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default] **at line 6 col 6** 

This warning appears in files with `.h` or `.hpp` extension, but not in those with `.cpp` extension.

ivankravets, 2016-06-23:
1. Try to use `.hpp` instead `.h`
2. `Menu: Rebuild Index C/C++`
3. Close Atom. Open it.

Does it work?

P.S: I've just checked it under Mac and it works very well.

ivankravets, 2016-06-23:
Please reopen if it doesn't work for you.

closed 2016-06-23 by ivankravets

frnhr, 2016-06-23:
Indeed it works! Thanks for the suggestion.

A side note: I'm striving to keep full compatibility with Arduino IDE when using PlatformIO / Atom, and apparently Arduino IDE knows about .hpp files too, so that's nice...