platformio/platformio-atom-ide #120: How to set speed of Serial Monitor to 74880 baud?

krzychb, 2016-06-05:
ESP8266 of ```platform = espressif ``` boots at 74880 baud providing debug messages on UART like:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
```

I was trying to set up speed of 74880 baud in Serial Monitor of PlatformIO IDE but do not see such possibility and have to use external serial terminal. 

[image]

Do I miss an option to somehow enter a custom speed in Serial Monitor or it is not supported?

orgkhnargh, 2016-06-05:
As a workaround you can put the line `<option value="74880">74880</option>` somewhere between lines 11 and 28 of the file `C:\Users\%USERNAME%\.atom\packages\platformio-ide\lib\serial-monitor\template.html`.

krzychb, 2016-06-05:
@orgkhnargh,

Thanks a lot for quick help with this issue!
Now I see why Atom is advertised as "A hackable text editor" :+1:

[image]

closed 2016-06-05 by krzychb