platformio/platformio-atom-ide #121: Failed to open database

martenbiehl, 2016-06-08:
After the update to the recent version I get two error messages when atom starts caused by platformio.

`Failed to open database
UnknownError: Internal error opening backing store for indexedDB.open.`

and
`An error occured during fetching recent projects
OpenFailedError: UnknownError Internal error opening backing store for indexedDB.open.`

This also causes platformio to open the home screen although I selected to not show it.

I installed platformio into my already existing atom installation.

IDE 1.2.2 | CLI 2.9.4

orgkhnargh, 2016-06-15:
Please open a developer console (click "View -> Developer -> Toggle Developer Tools") paste the string `indexedDB.deleteDatabase('platformio-ide')` and restart the editor.

martenbiehl, 2016-06-20:
Hi @orgkhnargh,

thanks that seems to have fixed the error message. However, platformio still does not respect the unchecked "Show PlatformIO Home screen on startup" option in the settings. Is that a separate issue?

orgkhnargh, 2016-06-20:
Hi @martenbiehl 

Have you been able to reproduce the issue on another computer? Are you sure it's unchecked? Run the string `atom.config.get('platformio-ide.showHomeScreen')` in a developer console and tell me what it will print back. Also press "Edit -> Config...`, wait until the config file opens and search for `showHomeScreen` option there. Is it `false` in the config (it is considered to be `true` if is not located in the file)?

Maybe this is a hardware issue? Did you encounter similar issues in other programs?

martenbiehl, 2016-06-22:
Hi @orgkhnargh,

thanks for the input. I removed my atom profile and reinstalled all packages. with a clean install everything works fine. maybe something in the update went wrong or and old package interfered.
Sorry to have wasted your time!

closed 2016-06-22 by martenbiehl

EmanuelSanchez, 2016-12-02:
Hello,
I have the same problem. When I open Atom after a while I get the error: "Failed to open a database. UnknownError: Internal error opening backing store for indexedDB.open." I uninstalled Atom and I deleted the .atom folder and I installed it again with the platformio package, but the error is still.

biazzotto, 2017-02-04:
@EmanuelSanchez try to delete, or rename, `~/.config/Atom`.