platformio/platformio-atom-ide #122: pio

majsi, 2016-06-09:
platformio : The term 'platformio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct
and try again.
At line:1 char:1
+ platformio update
+ ~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (platformio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

ivankravets, 2016-06-09:
Where do you see that output?

majsi, 2016-06-10:
I fix it with add to nevironments path
C:\Users\user\.atom\packages\platformio-ide\penv\Scripts

ivankravets, 2016-06-10:
Hm... IDE should append this to `PATH` automatically. Thanks for the answer.

closed 2016-06-10 by ivankravets