platformio/platformio-atom-ide #123: IDE package is active regardless or project?

polarathene, 2016-06-13:
This is my first package for Atom that adds IDE features around a specific environment. It's a fantastic project/package but I've noticed that it's not only active in my Arduino projects but any project (web/mobile/node/etc). It would be nice if the IDE package/features were only enabled when toggled or a suitable project is detected.

It doesn't seem to be mentioned in the docs, is the advised method to just disable the package and enable it whenever I work on a project suited for this package/IDE such as Arduino? Is there any package for Atom that you know of that might handle this better than manually every time I switch projects? If so perhaps mention it in the docs for Atom support.

ivankravets, 2016-06-14:
You can hide Toolbar: `Right Click > Hide Tool bar`.

polarathene, 2016-06-14:
@ivankravets Sure, but then I'd have to go show it from a menu or command pallete right? The platform.io tab opens on every project too, so it's more than just the toolbar that was bothering me. Disabling the package seems better way to go until I need it.

I haven't used the project manager package yet, but I think it might be a good way to go as it's meant to enable/disable packages based on your active project.

ivankravets, 2016-06-14:
>  I'd have to go show it from a menu or command pallete right?

You have a lot of options:
* Right Click (on the panel) > Hide Tool bar
* Menu: Packages > Tool Bar > Toggle
* `Ctrl+Alt+T

> The platform.io tab opens on every project too, so it's more than just the toolbar that was bothering me. 

Scroll down on PlatformIo Home page and uncheck "Show PlatformIO Home Screen on startup".

polarathene, 2016-06-15:
Thanks for the tip on disabling the Home Screen, turns out that if I enabled the package with it enabled it all open Atom windows would open the Home Screen tab.

I'm not presently using the Tool bar package besides with Platform.io, is it the same as the Atom tool bar package that is available without Platform.io, or would using that myself or with another IDE package with a toolbar conflict/merge?

ivankravets, 2016-06-15:
> is it the same as the Atom tool bar 

Yes, it is. We use the Tool bar API

closed 2016-06-15 by polarathene