platformio/platformio-atom-ide #124: [Linux] Can't see any terminal prompt

thunderbirdtr, 2016-06-13:
I have Fedora OS and made clean install system and atom. Then I install python tools via pip and install clang and everything else has installed. But when I click "terminal" I can't see anything on prompt It's just empty no "bash","zsh","sh" nothing is working at all. and here is screenshot;

[image]

Prompt just sit there but I can't type 

Thank you.

ivankravets, 2016-06-13:
1. Install `gcc` compiler and `npm`
2. Go to `~/.atom/packages/platformio-ide-terminal/node_modules/pty.js` and run `npm install`

thunderbirdtr, 2016-06-13:
I already have "gcc and "npm" and I ran it and still didn't have terminal.

ivankravets, 2016-06-13:
Could you provide an output of `npm install` in `node_modules/pty.js`?

thunderbirdtr, 2016-06-13:
`
> pty.js@0.3.0 install /home/thunderbirdtr/.atom/packages/platformio-ide-terminal/node_modules/pty.js
> node scripts/install.js

make: Entering directory '/home/thunderbirdtr/.atom/packages/platformio-ide-terminal/node_modules/pty.js/build'
  CXX(target) Release/obj.target/pty/src/unix/pty.o
  SOLINK_MODULE(target) Release/obj.target/pty.node
  COPY Release/pty.node
make: Leaving directory '/home/thunderbirdtr/.atom/packages/platformio-ide-terminal/node_modules/pty.js/build'

> pty.js@0.3.0 postinstall /home/thunderbirdtr/.atom/packages/platformio-ide-terminal/node_modules/pty.js
> node scripts/post-install.js

`

ivankravets, 2016-06-13:
Ok, let's try other hook

1. Change version of `nan` in `pty.js/package.json` from `2.0.5` to `2.3.5`
2. Remove `/pty.js/build` folder
3. Run again `npm install`
4. Restart Atom

thunderbirdtr, 2016-06-13:
did it and no terminal

thunderbirdtr, 2016-06-14:
Hello I fixed the problem myself and here is how I done ;

I installed atom-editor from repository which is installing seperating node packages so It's just atom binary and node packages coming from somewhere else. I checked my node and my atom so looks like I have tons of error with deps/unmed/ etc. So I delete atom from my system and install fresh one from "website" just rpm and done I also installed source-way "node"/"npm" and working better now. I have no-deps or unmed issue which is fix my problem. I just made simple test(with serial) and result is ; 

[image]

Sorry for trouble
Thank you for help 
(We can close this issue)

thunderbirdtr, 2016-06-14:
Oh btw before forget "Please add libncurses deps for upload" because It's needed for avr upload

ivankravets, 2016-06-14:
Do you mean that now you use an Atom package from the system repository?

thunderbirdtr, 2016-06-14:
Nope I'm using one from website(download and install standalone RPM) >> https://atom.io/download/rpm
I delete repository one from my system.

ivankravets, 2016-06-14:
1. Install the Atom using 
2. Install `platformio-ide` package from Atom Package Registry

Is it enough? Need to write some comment as final solution and place link on the top of the original comment. I hope that this info will be useful for other users.

Thanks.

thunderbirdtr, 2016-06-14:
Yes

I install atom from website and install package, install pip packages, install libncurses.so.5(depend on distro package name could be different so I type lib name) and done (Distro : fedora 24)

Thank you

ivankravets, 2016-06-14:
> install pip packages

Why do you need them? PlatformIO IDE has built-in PlatformIO CLI.

thunderbirdtr, 2016-06-14:
Ah yeah I notice that later yes. sorry

ivankravets, 2016-06-14:
Thanks. I've just updated your https://github.com/platformio/platformio-atom-ide/issues/124#issue-160031050

closed 2016-06-14 by ivankravets