platformio/platformio-atom-ide #125: Konstanten wie PI oder auch OUTPUT werden plötzlich nicht mehr erkannt?!

tennismichel, 2016-06-13:
Hallo zusammen, hoffe ihr könnt mir da weiterhelfen.

Hab beim "builden" von einem Testlauf zum nächsten plötzlich extrem viele Fehlermeldungen bekommen, die alle mehr oder weniger so aussehen wie die folgenden: 

.pioenvs/genuino101/FrameworkArduino/WCharacter.h:43:8: error: 'boolean' does not name a type
inline boolean isHexadecimalDigit(int c) __attribute__((always_inline));

.pioenvs/genuino101/FrameworkArduino/WCharacter.h:135:8: error: 'boolean' does not name a type
inline boolean isUpperCase(int c)

Zusätzlich sehe ich, dass Konstanten wie PI oder auch OUTPUT beim Pin zuweisen nicht erkannt werden. 

Am Code liegts definitiv nicht. Gleicher Code mit Platformio auf anderem Mac läuft ohne Probleme. Auch andere Platformio Projekte laufen bei mir jetzt nicht mehr. Bekomme überall die gleichen Fehlermeldungen. Neustarts, updates, Deinstallieren und wieder installieren hat alles nich geholfen.
hat jemand ne Ahnung was vlt noch helfen könnte.

Hab einen MAC mit der aktuellsten OS X Software. Nutze einen Genuino 101. Falls ihr sonst noch Infos braucht meldet euch. Vielen vielen Dank für eure Hilfe. 

Lg Michi

ivankravets, 2016-06-13:
Could you provide a code to reproduce this issue?

1. Does your project file have `*.cpp` extension?
2. Do you have this line in your project `#include <Arduino.h>`?

tennismichel, 2016-06-13:
Yes, I think I have both. Tried to upload the main and one of the header files. Not sure if it works.

[Quadro_main.txt](https://github.com/platformio/platformio-atom-ide/files/312727/Quadro_main.txt)
[RotorControl.txt](https://github.com/platformio/platformio-atom-ide/files/312728/RotorControl.txt)

ivankravets, 2016-06-13:
Please move issue to https://github.com/platformio/platformio/issues and fill all info

closed 2016-06-13 by ivankravets

ivankravets, 2016-06-14:
Moved issue https://github.com/platformio/platformio/issues/695