platformio/platformio-atom-ide #126: Soft wrap is passing the default value to  beautifying packages

peekpt, 2016-06-14:
I tried Beautifier Uncrustify and Prettify and the code line length gets truncated at 80 chars.

[Bug Gif](https://dl.dropboxusercontent.com/s/3kdcwsqndefrmx6/968153E2-35F6-444D-8362-8B59F10C7BDD-50330-00007DF4EAD8684A.gif?dl=0)

ivankravets, 2016-06-15:
Cool! How we can help you?

peekpt, 2016-06-15:
@ivankravets I posted an image gif with the issue, are you able to see it?
Atom has that software function to wrap code at desired line length but I think it's fixed at 80 chars. 
When you beautify code the line length truncates code at 80, even if you change it to 120 or so...
That vertical "end of row" representation line is working properly though.
I tried Atom-Beautify (using uncrustify) and Prettify (using clang) and it has the same behavior.
I don't know if it is truncated after beautifier do its job or is passing a variable to beautifier to truncate the code at that line length.

So the problem is I want a soft wrap function at 120 and I am getting 80. I am using developer version.

ivankravets, 2016-06-15:
But this issue is not related to `platformio-ide` package :( You need to contact with the authors of Atom-Beautify

peekpt, 2016-06-15:
Oh, I am sorry I tested with two different packages I thought it was a platformio-ide problem.

closed 2016-06-15 by ivankravets

peekpt, 2016-06-15:
I tried 3 code formatters and get the same issue, I don't see anything in the packages code referring to line-length code wrapping. :(, tried clang formatter, atom-beautifier and atom-prettify. Are you sure this is a 3d party bug? :/

orgkhnargh, 2016-06-15:
@peekpt 
> Are you sure this is a 3d party bug?

This is not a bug. You're trying to configure a `clang-format` through Atom setting interface, which is not possible (unless someone writes a program that can export Atom settings in `clang-format` format). There is [a guide](http://clang.llvm.org/docs/ClangFormatStyleOptions.html) on configuring the `clang-format`.

This particular issue can be resolved by creating a file named `.clang-format` in a project directory with the following content:

```
---
BasedOnStyle:  LLVM
ColumnLimit:     120
...
```

peekpt, 2016-06-15:
Solved, thanks.

on mac:
clang-format -style=llvm -dump-config > .clang-format

edit .clang-format and change ColumnLimit:  parameter