platformio/platformio-atom-ide #127: Endless installation loop

everm1nd, 2016-06-17:
I have a common problem with endless installation of PlatformIO IDE package. Package version is `platformio-ide@1.2.2`, so regarding a closed issues I saw it should be fixed already. And I don't have any `build`, `linter` and etc. directories in my home folder.

ivankravets, 2016-06-23:
Have you resolved this issue?

everm1nd, 2016-06-27:
Yep, it works now by some reason. Maybe since I've restarted my machine, but that's also strange

ivankravets, 2016-06-27:
Thanks a lot for the report. Please reopen this issue if you will have the same problem again.

closed 2016-06-27 by ivankravets

gnkarn, 2018-03-06:
this issue is still alive , with the new platfomio update , 
i can't not even use platform now , as it is in endless loop  asking for restart after update, but restart  only goes to the update ...

ivankravets, 2018-03-06:
@gnkarn what is your OS?

gnkarn, 2018-03-06:
it is mac OS X el cap

gnkarn, 2018-03-06:
[image]
after this , opens the restart button and then repeated the cycle again, and ..

gnkarn, 2018-03-06:
im not worried about the update itself, the problem is , that platformio is not working now ,home button is not working , and i can't open any project .

ivankravets, 2018-03-06:
Atom > View > Developer > Toggle Developer tools.

Do you see any errors?

gnkarn, 2018-03-06:
No i don't see any

gnkarn, 2018-03-06:
Uncaught (in promise) undefined  , in index.html

gnkarn, 2018-03-06:
this also, but not sure if is an error
User's Virtualenv: Error: spawn virtualenv ENOENT

gnkarn, 2018-03-06:
Error: Incompatible PIO Core 3.5.1
    at PlatformIOCoreStage._callee7$ (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2435:23)
    at tryCatch (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:832:40)
    at Generator.invoke [as _invoke] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1066:22)
    at Generator.prototype.(anonymous function) [as next] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:884:21)
    at step (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:191)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:361

ivankravets, 2018-03-06:
> Error: Incompatible PIO Core 3.5.1
http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

gnkarn, 2018-03-06:
Menu PlatformIO: Settings > PlatformIO IDE > **Use built-in PlatformIO Core**

does not appear on the menu

gnkarn, 2018-03-06:
the uninstall command gives these errors:

Proceed (y/n)? y
Exception:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/commands/uninstall.py", line 76, in run
    requirement_set.uninstall(auto_confirm=options.yes)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_set.py", line 346, in uninstall
    req.uninstall(auto_confirm=auto_confirm)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_install.py", line 754, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_uninstall.py", line 115, in remove
    renames(path, new_path)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/utils/__init__.py", line 267, in renames
    shutil.move(old, new)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 303, in move
    os.unlink(src)
OSError: [Errno 13] Permission denied: '/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio-3.5.1.dist-info/DESCRIPTION.rst'

gnkarn, 2018-03-06:
i think Menu PlatformIO: Settings > PlatformIO IDE > Use built-in PlatformIO Core , does not appear as the plaformio menu is  there but not working .

gnkarn, 2018-03-06:
is there other way to edit the plaformio config file  outside of atom ?

gnkarn, 2018-03-06:
i have successfully uninstalled  by using sudo .... , but  end result is exactly the same as before , something else you suggest to try ?, tks

gnkarn, 2018-03-06:
after uninstall , this error remains ( on developers window)

/Users/Gus-nuevo/.atom/packages/platformio-ide/lib/installer/manager.js:127 Error: Incompatible PIO Core 3.5.1
    at PlatformIOCoreStage._callee7$ (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2435:23)
    at tryCatch (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:832:40)
    at Generator.invoke [as _invoke] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1066:22)
    at Generator.prototype.(anonymous function) [as next] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:884:21)
    at step (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:191)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:361

**AND there are new errors now :** 

`/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2105 Error: ENOTEMPTY: directory not empty, rmdir '/Users/Gus-nuevo/.platformio/penv/lib/python2.7'
    at Object.fs.rmdirSync (fs.js:819:18)
    at rmkidsSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:356:25)
    at rmdirSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:334:7)
    at rimrafSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:304:9)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:342:5
    at Array.forEach (native)
    at rmkidsSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:341:26)
    at rmdirSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:334:7)
    at rimrafSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:304:9)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:342:5
    at Array.forEach (native)
    at rmkidsSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:341:26)
    at rmdirSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:334:7)
    at Function.rimrafSync [as sync] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:304:9)
    at Proxy.removeSync (/Users/Gus-nuevo/.atom/packages/linter-eslint/node_modules/fs-plus/lib/fs-plus.js:285:21)
    at PlatformIOCoreStage.cleanVirtualEnvDir (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2102:26)
    at PlatformIOCoreStage._callee8$ (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2485:22)
    at tryCatch (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:832:40)
    at Generator.invoke [as _invoke] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1066:22)
    at Generator.prototype.(anonymous function) [as next] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:884:21)
    at step (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:191)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:437
    at PlatformIOCoreStage.<anonymous> (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:99)
    at PlatformIOCoreStage.install (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2557:22)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/lib/installer/manager.js:91:38
    at Array.map (native)
    at InstallationManager.install (/Users/Gus-nuevo/.atom/packages/platformio-ide/lib/installer/manager.js:91:19)
    at PlatformIOIDEPackage.<anonymous> (/Users/Gus-nuevo/.atom/packages/platformio-ide/lib/main.js:84:18)
    at Generator.next (<anonymous>)
    at step (/Users/Gus-nuevo/.atom/packages/platformio-ide/lib/main.js:11:273)
cleanVirtualEnvDir @ /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2105
`

ivankravets, 2018-03-06:
Please remove `/Users/Gus-nuevo/.platformio/penv/` and restart IDE

gnkarn, 2018-03-06:
ok , 
it keeps on the same loop after restart .

now the other errors have disappeared , but **the original one remains**

/Users/Gus-nuevo/.atom/packages/platformio-ide/lib/installer/manager.js:127 Error: Incompatible PIO Core 3.5.1
    at PlatformIOCoreStage._callee7$ (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2435:23)
    at tryCatch (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:832:40)
    at Generator.invoke [as _invoke] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1066:22)
    at Generator.prototype.(anonymous function) [as next] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:884:21)
    at step (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:191)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:361

gnkarn, 2018-03-06:
should i delete this ?

(anonymous) | @ | /Users/Gus-nuevo/.atom/packages/platformio-ide/

ivankravets, 2018-03-06:
You have a lot of PIO Cores in a system. I don't know why did you install them manually. PIO IDE has built-in PIO Core. Please open system terminal (not IDE) and run `sudo pip uninstall platformio`. How did you install PIO Core? Using PIP or brew?

gnkarn, 2018-03-06:
i don't know either ...,
 i did already , and no changes ..

now  i receive this error 
**Cannot uninstall requirement platformio, not installed**

probably because i run the command before and was already uninstalled?

(pip)

ivankravets, 2018-03-06:
Ok, please check in system terminal `pio --version`. Do you see an output?

gnkarn, 2018-03-06:
at any folder ?

iMac-Gus-os10-ext-5:~ Gus-nuevo$ pio --version
-bash: pio: command not found

ivankravets, 2018-03-06:
Have you tried to remove `/Users/Gus-nuevo/.platformio/penv/` and restart IDE now?

gnkarn, 2018-03-06:
yes , same behavior ..
no changes

gnkarn, 2018-03-06:
now , have this errors

Error: ENOTEMPTY: directory not empty, rmdir '/Users/Gus-nuevo/.platformio/penv/lib/python2.7'
    at Object.fs.rmdirSync (fs.js:819:18)
    at rmkidsSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:356:25)
    at rmdirSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:334:7)
    at rimrafSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:304:9)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:342:5
    at Array.forEach (native)
    at rmkidsSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:341:26)
    at rmdirSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:334:7)
    at rimrafSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:304:9)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:342:5
    at Array.forEach (native)
    at rmkidsSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:341:26)
    at rmdirSync (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:334:7)
    at Function.rimrafSync [as sync] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/rimraf/rimraf.js:304:9)
    at Proxy.removeSync (/Users/Gus-nuevo/.atom/packages/linter-eslint/node_modules/fs-plus/lib/fs-plus.js:285:21)
    at PlatformIOCoreStage.cleanVirtualEnvDir (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2102:26)
    at PlatformIOCoreStage._callee8$ (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2485:22)
    at tryCatch (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:832:40)
    at Generator.invoke [as _invoke] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1066:22)
    at Generator.prototype.(anonymous function) [as next] (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:884:21)
    at step (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:191)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:437
    at PlatformIOCoreStage.<anonymous> (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1865:99)
    at PlatformIOCoreStage.install (/Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2557:22)
    at /Users/Gus-nuevo/.atom/packages/platformio-ide/lib/installer/manager.js:91:38
    at Array.map (native)
    at InstallationManager.install (/Users/Gus-nuevo/.atom/packages/platformio-ide/lib/installer/manager.js:91:19)
    at PlatformIOIDEPackage.<anonymous> (/Users/Gus-nuevo/.atom/packages/platformio-ide/lib/main.js:84:18)
    at Generator.next (<anonymous>)
    at step (/Users/Gus-nuevo/.atom/packages/platformio-ide/lib/main.js:11:273)
cleanVirtualEnvDir @ /Users/Gus-nuevo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2105

in addition to the alert:   Error: Incompatible PIO Core 3.5.1

ivankravets, 2018-03-06:
Do you have checked option with using development version of PIO Core in Atom / Settings / packages / PlatformIO IDE

gnkarn, 2018-03-06:
no i had not , i did and it seams to be working now , thank you !

ivankravets, 2018-03-07:
Happy coding with PlatformIO! 🤓