platformio/platformio-atom-ide #128: "Go to Declaration" (F3 hotkey) feature not working in Windows

DerTomm, 2016-06-20:
In contrast to Linux the "Go to Declaration" feature is not working in Windows environment.
My sample project is the Arduino Blink.ino equivalent Blink.cpp.

**Operating System**
Windows 10 64bit

**platformio.ini**
```
[env:nanoatmega328]
platform = atmelavr
framework = arduino
board = nanoatmega328
```

**Atom developers log output**
```
In file included from <stdin>:16:
C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:155:23: warning: unknown attribute __progmem__ ignored [-Wunknown-attributes]
extern const uint16_t PROGMEM port_to_mode_PGM[];
                      ^

autocomplete-clang.coffee:87 C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:109:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:95:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^
In file included from <stdin>:16:
C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:156:23: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint16_t PROGMEM port_to_input_PGM[];
                      ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:109:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:95:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^
In file included from <stdin>:16:
C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:157:23: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint16_t PROGMEM port_to_output_PGM[];
                      ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:109:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:95:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^
In file included from <stdin>:16:
C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:159:22: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint8_t PROGMEM digital_pin_to_port_PGM[];
                     ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:109:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:95:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^
In file included from <stdin>:16:
C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:161:22: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint8_t PROGMEM digital_pin_to_bit_mask_PGM[];
                     ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:109:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^

autocomplete-clang.coffee:87 C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:95:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^
In file included from <stdin>:16:
C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:162:22: warning: unknown attribute '__progmem__' ignored [-Wunknown-attributes]
extern const uint8_t PROGMEM digital_pin_to_timer_PGM[];
                     ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:109:17: note: expanded from macro 'PROGMEM'
#define PROGMEM __ATTR_PROGMEM__
                ^
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\avr/pgmspace.h:95:41: note: expanded from macro '__ATTR_PROGMEM__'
#define __ATTR_PROGMEM__ __attribute__((__progmem__))
                                        ^

autocomplete-clang.coffee:87 In file included from <stdin>:16:
In file included from C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:225:
In file included from C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino/USBAPI.h:27:
In file included from C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\util/delay.h:43:
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\util/delay_basic.h:108:5: error: invalid output constraint '=w' in asm
                : "=w" (__count)
                  ^

autocomplete-clang.coffee:87 In file included from <stdin>:16:
In file included from C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino\Arduino.h:225:
In file included from C:\Users\Thomas\.platformio\packages\framework-arduinoavr\cores\arduino/USBAPI.h:27:
C:\Users\Thomas\.platformio\packages\toolchain-atmelavr\avr\include\util/delay.h:95:3: warning: "Compiler optimizations disabled; functions from <util/delay.h> won't work as designed" [-W#warnings]
# warning "Compiler optimizations disabled; functions from <util/delay.h> won't work as designed"
  ^

autocomplete-clang.coffee:87 7 warnings and 1 error generated.
```

snowbizz84, 2016-10-10:
Any progress in this bug?

drepou, 2016-10-21:
same issue for me

vitoralvimb, 2016-10-26:
Same Error for me. I am using the latest version...

`In file included from <stdin>:7:
In file included from C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Arduino.h:27:
In file included from C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\stdlib.h:18:
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:185:36: error: C++ requires a type specifier for all declarations
  _READ_WRITE_RETURN_TYPE _EXFNPTR(_read, (struct _reent *, _PTR,
                                   ^

autocomplete-clang.coffee:91 C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:185:44: error: C++ requires a type specifier for all declarations
  _READ_WRITE_RETURN_TYPE _EXFNPTR(_read, (struct _reent *, _PTR,
                                           ^
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:187:36: error: C++ requires a type specifier for all declarations
  _READ_WRITE_RETURN_TYPE _EXFNPTR(_write, (struct _reent *, _PTR,
                                   ^
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:187:45: error: C++ requires a type specifier for all declarations
  _READ_WRITE_RETURN_TYPE _EXFNPTR(_write, (struct _reent *, _PTR,
                                            ^
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:189:20: error: C++ requires a type specifier for all declarations
  _fpos_t _EXFNPTR(_seek, (struct _reent *, _PTR, _fpos_t, int));
                   ^
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:189:28: error: C++ requires a type specifier for all declarations
  _fpos_t _EXFNPTR(_seek, (struct _reent *, _PTR, _fpos_t, int));
                           ^

autocomplete-clang.coffee:91 C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:190:16: error: C++ requires a type specifier for all declarations

autocomplete-clang.coffee:91   int _EXFNPTR(_close, (struct _reent *, _PTR));
               ^

autocomplete-clang.coffee:91 C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:190:25: error: C++ requires a type specifier for all declarations
  int _EXFNPTR(_close, (struct _reent *, _PTR));
                        ^

autocomplete-clang.coffee:91 C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:597:17: error: C++ requires a type specifier for all declarations
  void _EXFNPTR(__cleanup, (struct _reent *));
                ^

autocomplete-clang.coffee:91 C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\sys/reent.h:597:29: error: C++ requires a type specifier for all declarations
  void _EXFNPTR(__cleanup, (struct _reent *));
                            ^

autocomplete-clang.coffee:91 In file included from <stdin>:7:
In file included from C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Arduino.h:36:
C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/stdlib_noniso.h:29:5: error: conflicting types for 'atoi'
int atoi(const char *s);
    ^
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\stdlib.h:70:12: note: previous declaration is here

autocomplete-clang.coffee:91 int     _EXFUN(atoi,(const char *__nptr));
               ^
In file included from <stdin>:7:
In file included from C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Arduino.h:36:
C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/stdlib_noniso.h:31:6: error: conflicting types for 'atol'
long atol(const char* s);
     ^
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\stdlib.h:72:13: note: previous declaration is here
long    _EXFUN(atol,(const char *__nptr));
               ^
In file included from <stdin>:7:
In file included from C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Arduino.h:36:

autocomplete-clang.coffee:91 C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/stdlib_noniso.h:33:8: error: conflicting types for 'atof'
double atof(const char* s);
       ^
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\stdlib.h:66:15: note: previous declaration is here
double  _EXFUN(atof,(const char *__nptr));
               ^

autocomplete-clang.coffee:91 In file included from <stdin>:7:
[... 38 lines cut ...]
    return (isalpha(c) == 0 ? false : true);
            ^~~~~~~
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\ctype.h:9:12: note: candidate function not viable: requires 0 arguments, but 1 was provided
int _EXFUN(isalpha, (int __c));
           ^
In file included from <stdin>:7:
In file included from C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:242:
C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WCharacter.h:65:13: error: no matching function for call to 'isblank'
    return (isblank(c) == 0 ? false : true);
            ^~~~~~~
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\ctype.h:23:12: note: candidate function not viable: requires 0 arguments, but 1 was provided
[... cut ...]
C:\Users\Vitor\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WCharacter.h:70:13: error: no matching function for call to 'iscntrl'
    return (iscntrl(c) == 0 ? false : true);
            ^~~~~~~
C:\Users\Vitor\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\ctype.h:10:12: note: candidate function not viable: requires 0 arguments, but 1 was provided

autocomplete-clang.coffee:91 int _EXFUN(iscntrl, (int __c));
           ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]

autocomplete-clang.coffee:91 2 warnings and 20 errors generated.`

obertr0n, 2016-12-19:
I have seen that this issue is still unresolved for Windows uses.
I think that it's a powerful tool for developers and shouldn't be neglected.
Can you please provide a date for the fix?

Spongman, 2017-04-01:
Does anyone have any idea why this is happening? Is clang for Windows out of date? Are the wrong flags getting passed?

ivankravets, 2017-04-30:
Resolved in the latest beta

closed 2017-04-30 by ivankravets