platformio/platformio-atom-ide #129: Serial monitor leaves python processes hanging

exstac, 2016-06-23:
Every upload seems to leave a python2.7 process around working at 100% CPU. The only way I've found to get rid of them is to issue a killall python2.7 from the terminal.

The board I'm working with is a NodeMCU ESP-12E if that makes any difference.

ivankravets, 2016-06-24:
What is your OS?

exstac, 2016-06-25:
I'm running Mac OS X El Capitan, latest updates

ivankravets, 2016-06-25:
In the title you say about "Serial Monitor" but in the post about "Every upload seems to leave a python2.7 process around working at 100% CPU.".

Do you have problems with uploading or serial monitor?

exstac, 2016-06-25:
In trying to reproduce the issue I found that it only happened when uploading. The serial monitor (which opens after the upload is complete) seemed to have been a red herring.

ivankravets, 2016-06-25:
1. Open PlatformIO IDE Terminal
2. Run `platformio run -t upload`

Do you have the same problem?

exstac, 2016-06-25:
No, that uploads without leaving any processes hanging.
I grabbed the output from ps and it is actually the serial monitor that is hanging around:
```
USER              PID  %CPU %MEM      VSZ    RSS   TT  STAT STARTED      TIME COMMAND
oscar           55615 100.0  0.1  2537008   6664   ??  R     3:03PM   2:02.76 /Users/oscar/.atom/packages/platformio-ide/penv/bin/python2.7 /Users/oscar/.atom/packages/platformio-ide/penv/bin/pio -f -c atom serialports monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
oscar           55398 100.0  0.1  2535984   5728   ??  R     3:02PM   2:41.36 /Users/oscar/.atom/packages/platformio-ide/penv/bin/python2.7 /Users/oscar/.atom/packages/platformio-ide/penv/bin/pio -f -c atom serialports monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
oscar           55831   0.0  0.1  2538812   5132 s004  S+    3:03PM   0:00.27 /Users/oscar/.atom/packages/platformio-ide/penv/bin/python2.7 /Users/oscar/.atom/packages/platformio-ide/penv/bin/pio -f -c atom serialports monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
```

I have the IDE open with the serial monitor running (working). After quitting the IDE the last process in the list above goes up to 100% like the first two:
```
USER              PID  %CPU %MEM      VSZ    RSS   TT  STAT STARTED      TIME COMMAND
oscar           55398 100.0  0.1  2535984   4728   ??  R     3:02PM   4:23.60 /Users/oscar/.atom/packages/platformio-ide/penv/bin/python2.7 /Users/oscar/.atom/packages/platformio-ide/penv/bin/pio -f -c atom serialports monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
oscar           55831 100.0  0.1  2534960   6364   ??  R     3:03PM   0:35.15 /Users/oscar/.atom/packages/platformio-ide/penv/bin/python2.7 /Users/oscar/.atom/packages/platformio-ide/penv/bin/pio -f -c atom serialports monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
oscar           55615 100.0  0.1  2537008   6660   ??  R     3:03PM   3:44.99 /Users/oscar/.atom/packages/platformio-ide/penv/bin/python2.7 /Users/oscar/.atom/packages/platformio-ide/penv/bin/pio -f -c atom serialports monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
```

exstac, 2016-06-25:
I also want to add that each deploy (with the serial monitor open) will leave one python2.7 process running at 100% CPU

ivankravets, 2016-06-26:
Do you have specific project? Could you share it here? Do you have the same problems with "Blink" project?

exstac, 2016-06-27:
This program triggers it:
platformio.ini:
```
[env:nodemcuv2]
platform = espressif
framework = arduino
board = nodemcuv2
```

main.cpp:
```
#include <Arduino.h>

void setup() {
  pinMode(16, OUTPUT);
  Serial.begin(115200);
  delay(500);
  Serial.println("setup done");
}

void loop() {
  Serial.println("loop");
  digitalWrite(16, HIGH);
  delay(1000);
  digitalWrite(16, LOW);
  delay(1000);
}
```

ivankravets, 2016-06-27:
I have the same OS, the same board, and the same configuration. Tried to play with your project and didn't find any problems :(

Could you provide TeamViwer session for debug? Thanks.

ivankravets, 2016-07-10:
Do you still have this problem?

ivankravets, 2016-07-18:
Please reopen if you still need a help

mrkaroshi, 2017-03-24:
Is there a fix for this yet? I'm having to close my Atom application every hour or so so my computer doesn't run on hot all day.

ivankravets, 2017-03-25:
Guys, I need more info here:

1. Try to open PlatformIO IDE Terminal and run `pio device monitor` manually.
2. Check that process exists
3. Don't press `Ctrl+C` but use 'close' (destroy) icon in Terminal
4. Does process exist?

It looks that PIO Terminal doesn't destroy child process correctly.

ivankravets, 2017-06-01:
Please reopen if you still need help

closed 2017-06-01 by ivankravets

eethann, 2017-09-20:
I am having this issue as well. Versions are Atom 1.20 and PlatformIO 2.6.0. The issue appears to be related to uploading not terminating existing monitor processes. I can reliably reproduce with the following:

1. Open a serial monitor pane.
2. Upload to device (in my case a Teensy 3.2)
3. The serial monitor pane closes, build and upload commence, then a new serial monitor pane comes up

If I inspect my processes in a terminal with `ps aux | grep processio` I see multiple monitoring processes running, even though only one monitoring pane is shown in the video.

If I open the monitor in the PIO terminal via command line this does not happen, I don't think. (not tested rigorously).

FWIW I'm on Yosemite, but I see others are seeing this on more recent OS versions.

pruchai, 2018-01-26:
I have having the same issue. When the serial monitor is running and build initiated it closes the serial monitor window, but does not kill the process. After a few times I end up with a bunch of python processes left behind and fans going crazy, so after each build I have to do

`pkill -f python2.7`