platformio/platformio-atom-ide #134: platformio internal command

zjbjbjbz, 2016-07-01:
Hi,

I have a trouble when build. the ide says me 'platformio' is not recognized as an internal command.
I try to modify path envioment but without success.
Somebody can help me.
PS : i'm on win 7 with python 2.7 and aTom version 1.8 
Thks

ivankravets, 2016-07-10:
Looks like this issue was fixed in https://github.com/platformio/platformio-atom-ide/commit/89a636dce129394dd4f7265b861dd2e3afb88011

Could you try to replace `C:/Users/%USER%/.atom/packages/platformio-ide/lib/maintenance.js` with [this](https://raw.githubusercontent.com/platformio/platformio-atom-ide/develop/lib/maintenance.js)?

Did it help you?

closed 2016-07-11 by ivankravets

ivankravets, 2016-07-12:
PlatformIO IDE 1.3.1 has been released. Please update it.

zjbjbjbz, 2016-07-12:
Hi
I add .atom\packages\platformio-ide\penv\Scripts in path environment, and now it works.
Thx

ivankravets, 2016-07-12:
Try to remove it from path environment and install the latest 1.3.1 version. It should work. Thanks.

GregTerrell, 2018-09-18:
I am seeing this issue as well (Windows 10, VSCode 1.27.2) and the PlatformIO install link within VSCode says 1.2.0.  Hows does one get the latest 1.3.1?

The download link says "Download and install official Microsoft's Visual Studio Code, PlatformIO IDE is built on top of it".

Brand new install, patiently waited for install task to finish, this is the 1st attempt to create a (arduino) project for development under Visual Studio (mainly due to existing VSTS workflows).  Evaluating if my company should switch from competitive IDE.
Thanks for your help,
Greg

ivankravets, 2018-09-19:
@GregTerrell this is repository is intended for discounted PlatformIO IDE for Atom. If you have any issues with PlatformIO IDE for VSCode, please file a request here https://github.com/platformio/platformio-vscode-ide

We will help ASAP. Thanks.