platformio/platformio-atom-ide #135: Platformio doesn't generate EEPROM HEX

Trel725, 2016-07-02:
If using placing constants into EEPROM memory, it is necessary to generate appropriate HEX file to program EEPROM. Platformio doesn't do that.

From Atmel reference:
> If we specify the EEPROM segment, we can generate a .hex file that can be used to program the EEPROM:
$ avr-objcopy -j .eeprom --change-section-lma .eeprom=0 -O ihex demo.elf demo_eeprom.hex

ivankravets, 2016-07-04:
Please move this issue to PlatformIO CLI repository https://github.com/platformio/platformio/issues

ivankravets, 2016-07-04:
You need to call `pio run -t uploadeep` from the PlatformIO IDE Terminal.

Does it work for you?

Trel725, 2016-07-04:
Yes it does. I am sorry, but it is not obvious from CLI help.

closed 2016-07-04 by Trel725

zharovdv, 2023-03-25:
> Is there a way to generate the eep file without uploading it as well?

My way is:

eeprom.py
```
from os.path import join

Import("env")

env.Depends(join("$BUILD_DIR", "${PROGNAME}.hex"), join("$BUILD_DIR", "${PROGNAME}.eep"))
```

platformio.ini

`extra_scripts = eeprom.py`