platformio/platformio-atom-ide #137: Uncaught Error: ENOENT: no such file or directory, lstat 'C:\Users\htko8\AppData\Local\atom\app-1...

htko89, 2016-07-12:
Not sure what is happening, my platformio package on atom updated to the latest version today and threw this error.

[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.8.0
**System**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.3.1

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, lstat 'C:\Users\htko8\AppData\Local\atom\app-1.8.0\atom:'

```
At ATOM_SHELL_ASAR.js:0

Error: ENOENT: no such file or directory, lstat 'C:\Users\htko8\AppData\Local\atom\app-1.8.0\atom:'
    at Error (native)
    at fs.lstatSync (fs.js:887:18)
    at Object.fs.lstatSync (ATOM_SHELL_ASAR.js:229:16)
    at Object.realpathSync (fs.js:1532:21)
    at Object.fs.realpathSync (ATOM_SHELL_ASAR.js:304:29)
    at new PlatformIOBuildProvider (C:/Users/htko8/.atom/packages/platformio-ide/lib/build-provider.js:32:19)
    at C:/Users/htko8/.atom/packages/build/lib/target-manager.js:67:22
    at Array.map (native)
    at C:/Users/htko8/.atom/packages/build/lib/target-manager.js:67:10
    at Array.map (native)
    at TargetManager.refreshTargets (C:/Users/htko8/.atom/packages/build/lib/target-manager.js:58:39)
    at C:/Users/htko8/.atom/packages/build/lib/target-manager.js:18:12
    at Function.module.exports.Emitter.simpleDispatch (C:\Users\htko8\AppData\Local\atom\app-1.8.0\resources\app.asar\node_modules\event-kit\lib\emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (C:\Users\htko8\AppData\Local\atom\app-1.8.0\resources\app.asar\node_modules\event-kit\lib\emitter.js:125:28)
    at Project.module.exports.Project.addPath (C:\Users\htko8\AppData\Local\atom\app-1.8.0\resources\app.asar\src\project.js:263:29)
    at AtomEnvironment.module.exports.AtomEnvironment.openLocations (C:\Users\htko8\AppData\Local\atom\app-1.8.0\resources\app.asar\src\atom-environment.js:1120:26)
    at EventEmitter.outerCallback (C:\Users\htko8\AppData\Local\atom\app-1.8.0\resources\app.asar\src\application-delegate.js:244:18)
    at emitThree (events.js:102:20)
    at EventEmitter.emit (events.js:175:7)
```

### Commands

```
     -0:03.3.0 application:show-settings (atom-text-editor.editor)
```

### Config

```json
{
  "platformio-ide": {
    "showHomeScreen": false
  }
}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.9.4
build, v0.65.0
busy, v0.5.0
language-ini, v1.16.0
linter, v1.11.14
linter-gcc, v0.6.14
platformio-ide, v1.3.1
platformio-ide-terminal, v2.1.0
sync-settings, v0.7.2
tool-bar, v1.0.0

# Dev
No dev packages
```

ivankravets, 2016-07-12:
Try to close and open Atom. Did it help?

htko89, 2016-07-12:
On one of my computers it helped, On the other one, it continues to throw the same error. They have the same programs and packages installed.

htko89, 2016-07-12:
It seems to occur when moving in and out of the settings window, as it happens when I move from an open file tab to an already opened settings tab.

ivankravets, 2016-07-13:
Does this file `C:\Users\htko8\AppData\Local\atom\app-1.8.0\atom` exist?

ivankravets, 2016-07-13:
Try:

1. Uninstall Atom using "OS Control Panel > Uninstall Atom"
2. Remove 2 folders: `Users/%NAME%/.atom` and `Users/%NAME%/AppData/Local/atom`
3. Follow instructions here http://docs.platformio.org/en/latest/ide/atom.html#ide-installation . Note! Don't use our bundles. We are working on the new bundles that will be based on Atom 1.8.0

ivankravets, 2016-07-13:
That is bug. Thanks! We will fix it as soon as possible.

ivankravets, 2016-07-13:
Please update PlatformIO IDE to 1.3.2

closed 2016-07-13 by ivankravets

htko89, 2016-07-14:
Thank you for the hard work :+1:

commit 32d68949 (2016-07-13): Hotfix for "Uncaught Error" issue when "Settings" or "About" tabs are opened // Resolve #137