platformio/platformio-atom-ide #138: 'platformio' is not recognized as an internal or external command, operable program or batch file.

captainwasabi, 2016-07-13:
Updated platformio

Click on build from the toolbar.

'platformio' is not recognized as an internal or external command,
operable program or batch file.

However, all the commands work fine in the shell.

Windows 10

ivankravets, 2016-07-13:
Do you use the latest 1.3.1 version of IDE?

It should be fixed here https://github.com/platformio/platformio-atom-ide/issues/134

captainwasabi, 2016-07-13:
yes, just upgraded this morning.  There were some issues and I had to uninstall and reinstall.

ivankravets, 2016-07-13:
>  There were some issues 

We know about these issues and they are fixed in this 1.3.1. You will not see these errors for the next update

----

Did it work before with 1.3.0?

captainwasabi, 2016-07-13:
also now getting this trying to open a serial monitor:

io -f -c atom serialports monitor --port COM7
could not open port u'COM7': could not open port u'COM7': WindowsError(5, 'Access is denied.')

captainwasabi, 2016-07-13:
Ivan, I am on 1.3.1, that is what I installed.  Not sure what version I was on before.  I just installed it last week, so probably 1.3.0, and it did work.

ivankravets, 2016-07-13:
Could you provide me [TeamViewer](https://www.teamviewer.com) session for debug?

captainwasabi, 2016-07-13:
check your personal email

captainwasabi, 2016-07-13:
solved. Atom wasn't upgrading to latest.  I'll leave this open for your notes.

ivankravets, 2016-07-13:
Thanks, we are working on the new installers for PlatformIO IDE with the latest Atom 1.8.0

commit cabc76cf (2016-07-21): Fix issue "‘platformio’ is not recognized as an internal or external command" for Windows OS // Resolve #138
closed 2016-07-13 by ivankravets