platformio/platformio-atom-ide #139: Uncaught Error: ENOENT: no such file or directory, lstat '/atom:'

gopalmetro, 2016-07-13:
[Enter steps to reproduce below:]

1. Clicked Atom->About Atom

**Atom Version**: 1.8.0
**System**: Mac OS X 10.11.5
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.3.1

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, lstat '/atom:'

```
At ATOM_SHELL_ASAR.js:0

Error: ENOENT: no such file or directory, lstat '/atom:'
    at Error (native)
    at fs.lstatSync (fs.js:887:18)
    at Object.fs.lstatSync (ATOM_SHELL_ASAR.js:229:16)
    at Object.realpathSync (fs.js:1532:21)
    at Object.fs.realpathSync (ATOM_SHELL_ASAR.js:304:29)
    at /Users/gopalmetro/.atom/packages/platformio-ide/lib/utils.js:132:55
    at Array.map (native)
    at getActiveProjectPath (/Users/gopalmetro/.atom/packages/platformio-ide/lib/utils.js:132:41)
    at Object.highlightActiveProject (/Users/gopalmetro/.atom/packages/platformio-ide/lib/maintenance.js:252:13)
    at doHighlight (/Users/gopalmetro/.atom/packages/platformio-ide/lib/main.js:235:43)
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:125:28)
    at stoppedChangingActivePaneItemCallback (/Applications/Atom.app/Contents/Resources/app.asar/src/pane-container.js:350:36)
```

### Commands

```
```

### Config

```json
{}
```

### Installed Packages

```coffee
# User
autocomplete-clang, v0.9.4
build, v0.65.0
busy, v0.5.0
language-ini, v1.16.0
linter, v1.11.14
linter-gcc, v0.6.14
platformio-ide, v1.3.1
platformio-ide-terminal, v2.1.1
tool-bar, v1.0.0

# Dev
No dev packages
```

gopalmetro, 2016-07-13:
I installed a fresh copy of Atom, then installed the PlatformIO package.  When I clicked the Atom->About Atom menu item, this error popped up.

ivankravets, 2016-07-13:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/137

closed 2016-07-13 by ivankravets