platformio/platformio-atom-ide #140: Use electron-builder to generate platformio-atom-ide

probonopd, 2016-07-17:
Use [electron-builder](https://github.com/electron-userland/electron-builder) to generate platformio-atom-ide.

It can build binaries for Windows, macOS, and Linux (AppImage and deb).

ivankravets, 2016-07-17:
Have you tested it with PlatformIO IDE?

probonopd, 2016-07-17:
No since I am still new to it myself, but I am optimistic that @develar from the electron-builder team would be willing to have a look if you're interested.

develar, 2016-07-18:
I see that PlatformIO IDE uses https://raw.githubusercontent.com/atom/ci/master/build-package.sh to build as an "atom package". So, electron-builder cannot be used to build AppImage.

probonopd, 2016-07-18:
@develar "cannot be used" as in "...without changing from build-package.sh to electron-builder"? What would be the side effects?