platformio/platformio-atom-ide #142: Causes Atom to crash on Ubuntu 16.04 LTS

AnthonyJAttard, 2016-07-20:
OptiPlex-GX620
Intel® 945G
Program froze on launch and I had to remove the package from the files on disk before atom would run.

pgebhard, 2016-07-22:
This doesn't really provide much detail. Were you able to reproduce the issue? If so, who about with a clean install of Atom and trying to install Platformio again?

AnthonyJAttard, 2016-07-22:
The problem is just on this linux PC. The issue is repeatable even after a clean install. Atom works fine without PlatformIO.

ivankravets, 2016-07-22:
PlatformIO IDE 1.3.5 has been released today. Did you try it?

ivankravets, 2016-07-27:
Does clean version of http://atom.io work for you? If yes, please reopen this issue.

closed 2016-07-27 by ivankravets

AnthonyJAttard, 2016-07-27:
Problem solved after the new update was installed. Thank you for your help and keep up the good work!