platformio/platformio-atom-ide #146: MA OS X : PlatformIO: Unable to create a virtualenv.

schurrer, 2016-07-27:
When I install PlatformIO IDE and run it I receive error : 

PlatformIO: Unable to create a virtualenv.
PlatformIO Home
PlatformIO: Unable to create a virtualenv.
Traceback (most recent call last):
File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/virtualenv-root116627-26389-130v5t3/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 2316, in 
main()
File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/virtualenv-root116627-26389-130v5t3/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 708, in main
symlink=options.symlink)
File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/virtualenv-root116627-26389-130v5t3/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 941, in create_environment
download=download,
File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/virtualenv-root116627-26389-130v5t3/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 897, in install_wheel
call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/virtualenv-root116627-26389-130v5t3/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 792, in call_subprocess
% (cmd_desc, proc.returncode))
OSError: Command /Users/schurjar/.ato...e/penv/bin/python2.7 - setuptools pip wheel failed with error code 1

-------------------------------
ProductName: Mac OS X
ProductVersion: 10.11.6
BuildVersion: 15G19a

virtualenv is installed by pip and functional in command line.

pip list | grep virtu*
virtualenv (15.0.2)

Python 2.7.12 |Anaconda 2.1.0 (x86_64)| (default, Jul  2 2016, 17:43:17) 
[GCC 4.2.1 (Based on Apple Inc. build 5658) (LLVM build 2336.11.00)] on darwin

ivankravets, 2016-07-28:
Please run this command and restart IDE
```
virtualenv ~/.atom/packages/platformio-ide/penv
```

ivankravets, 2016-07-28:
@schurrer did it help?

schurrer, 2016-07-28:
Yes, it helps. I closed the issue.

Thanks. Jaroslav

ivankravets, 2016-07-28:
It seems that problem is caused by "anaconda" :( Would like to debug it? PlatformIO IDE Installer tries to make virtual environment automatically.

schurrer, 2016-07-28:
Of course I can. Send me how and I will do it.

ivankravets, 2016-07-28:
1. Download https://pypi.python.org/packages/source/v/virtualenv/virtualenv-15.0.1.tar.gz
2. Unpack it to `/tmp` or somewhere
3. `mkdir /tmp/virtualenv-root`
4. `cd /tmp/virtualenv-15.0.1/`
5. `python setup.py install --root /tmp/virtualenv-root/`
6. `python /tmp/virtualenv-root/Library/Python/2.7/site-packages/virtualenv.py /tmp/pio_penv`

Does it work for you?

ivankravets, 2016-07-28:
@schurrer I've just improved installer. Could you try it?

1. Replace `~/.atom/packages/platformio-ide/lib/install/command.js` with [this](https://raw.githubusercontent.com/platformio/platformio-atom-ide/develop/lib/install/command.js).
2. Remove `~/.atom/packages/platformio-ide/.cache` and `~/.atom/packages/platformio-ide/penv` folders if exists
3. Restart IDE.

ivankravets, 2016-07-30:
@schurrer do you have any news?

schurrer, 2016-07-30:
> Download https://pypi.python.org/packages/source/v/virtualenv/virtualenv-15.0.1.tar.gz <https://pypi.python.org/packages/source/v/virtualenv/virtualenv-15.0.1.tar.gz>
> Unpack it to /tmp or somewhere
> mkdir /tmp/virtualenv-root
> cd /tmp/virtualenv-15.0.1/
> python setup.py install --root /tmp/virtualenv-root/
> python /tmp/virtualenv-root/Library/Python/2.7/site-packages/virtualenv.py /tmp/pio_penv
> Does it work for you?

Step 6 doesn’t work. There is no such directory 	• /Library/Python/2.7/

Step 5 output:

python setup.py install --root ~/TMP/tst/virtualenv-root/
running install
running build
running build_py
running install_lib
creating /Users/schurjar/TMP/tst/virtualenv-root/Users
creating /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar
creating /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda
creating /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda
creating /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib
creating /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7
creating /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages
copying build/lib/virtualenv.py -> /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages
creating /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv_support
copying build/lib/virtualenv_support/__init__.py -> /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv_support
copying build/lib/virtualenv_support/argparse-1.4.0-py2.py3-none-any.whl -> /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv_support
copying build/lib/virtualenv_support/pip-8.1.1-py2.py3-none-any.whl -> /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv_support
copying build/lib/virtualenv_support/setuptools-20.3-py2.py3-none-any.whl -> /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv_support
copying build/lib/virtualenv_support/wheel-0.29.0-py2.py3-none-any.whl -> /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv_support
byte-compiling /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py to virtualenv.pyc
byte-compiling /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv_support/__init__.py to __init__.pyc
running install_egg_info
running egg_info
writing pbr to virtualenv.egg-info/pbr.json
writing virtualenv.egg-info/PKG-INFO
writing top-level names to virtualenv.egg-info/top_level.txt
writing dependency_links to virtualenv.egg-info/dependency_links.txt
writing entry points to virtualenv.egg-info/entry_points.txt
reading manifest file 'virtualenv.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no previously-included files matching '*' found under directory 'docs/_templates'
warning: no previously-included files matching '*' found under directory 'docs/_build'
writing manifest file 'virtualenv.egg-info/SOURCES.txt'
Copying virtualenv.egg-info to /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv-15.0.1-py2.7.egg-info
running install_scripts
Installing virtualenv script to /Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/bin

This folder structure is created after step 5:

ls -R  ~/TMP/tst/virtualenv-root/ 
Users

/Users/schurjar/TMP/tst/virtualenv-root//Users:
schurjar

/Users/schurjar/TMP/tst/virtualenv-root//Users/schurjar:
anaconda

/Users/schurjar/TMP/tst/virtualenv-root//Users/schurjar/anaconda:
anaconda

/Users/schurjar/TMP/tst/virtualenv-root//Users/schurjar/anaconda/anaconda:
bin	lib

/Users/schurjar/TMP/tst/virtualenv-root//Users/schurjar/anaconda/anaconda/bin:
virtualenv

/Users/schurjar/TMP/tst/virtualenv-root//Users/schurjar/anaconda/anaconda/lib:
python2.7

/Users/schurjar/TMP/tst/virtualenv-root//Users/schurjar/anaconda/anaconda/lib/python2.7:
site-packages

/Users/schurjar/TMP/tst/virtualenv-root//Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages:
virtualenv-15.0.1-py2.7.egg-info	virtualenv.pyc
virtualenv.py				virtualenv_support
[... 21 lines cut ...]
  Complete output from command /Users/schurjar/TMP/tst/pio_penv/bin/python - setuptools pip wheel:
  Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "/Users/schurjar/anaconda/anaconda/lib/python2.7/tempfile.py", line 32, in <module>
    import io as _io
  File "/Users/schurjar/anaconda/anaconda/lib/python2.7/io.py", line 51, in <module>
    import _io
ImportError: dlopen(/Users/schurjar/TMP/tst/pio_penv/lib/python2.7/lib-dynload/_io.so, 2): Symbol not found: __PyCodecInfo_GetIncrementalDecoder
  Referenced from: /Users/schurjar/TMP/tst/pio_penv/lib/python2.7/lib-dynload/_io.so
  Expected in: flat namespace
 in /Users/schurjar/TMP/tst/pio_penv/lib/python2.7/lib-dynload/_io.so
----------------------------------------
...Installing setuptools, pip, wheel...done.
Traceback (most recent call last):
  File "/Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 2316, in <module>
    main()
  File "/Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 708, in main
    symlink=options.symlink)
  File "/Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 941, in create_environment
    download=download,
  File "/Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 897, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/Users/schurjar/TMP/tst/virtualenv-root/Users/schurjar/anaconda/anaconda/lib/python2.7/site-packages/virtualenv.py", line 792, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /Users/schurjar/TMP/tst/pio_penv/bin/python - setuptools pip wheel failed with error code 1

ivankravets, 2016-07-30:
Thanks!

Is this https://github.com/platformio/platformio-atom-ide/issues/146#issuecomment-235995481 solution also doesn't work?

schurrer, 2016-07-30:
Output: 

PlatformIO: Unable to create a virtualenv.
Traceback (most recent call last):
  File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/d-116630-2415-eweqo0/virtualenv-15.0.1/virtualenv.py", line 2316, in <module>
    main()
  File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/d-116630-2415-eweqo0/virtualenv-15.0.1/virtualenv.py", line 708, in main
    symlink=options.symlink)
  File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/d-116630-2415-eweqo0/virtualenv-15.0.1/virtualenv.py", line 941, in create_environment
    download=download,
  File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/d-116630-2415-eweqo0/virtualenv-15.0.1/virtualenv.py", line 897, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/var/folders/k2/yzdy3gk912d9z2jywkbm7qsm0000gn/T/d-116630-2415-eweqo0/virtualenv-15.0.1/virtualenv.py", line 792, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /Users/schurjar/.ato...e/penv/bin/python2.7 - setuptools pip wheel failed with error code 1

ivankravets, 2016-07-30:
Hm... I can't realize, is this problem linked with `virtualenv` or your system environment?

Could you try this https://github.com/platformio/platformio-atom-ide/issues/146#issuecomment-235973537 , but use [14.0.6 version](https://github.com/pypa/virtualenv/releases) instead?

closed 2016-07-31 by ivankravets

yzliao, 2016-09-06:
I think this is a quick fix (https://github.com/conda/conda/issues/1367): downgrade python to 2.7.9, e.g., `conda install python=2.7.9`

commit 914aca03 (2016-07-31): Improve PlatformIO IDE Installer for Linux Arch and macOS // Resolve #146