platformio/platformio-atom-ide #148: 'fs-extra' sub-module issue

sigxcpu76, 2016-07-29:
Atom 1.8.0, latest platformio-ide package.

This one is bugging me for few days now and I don't understand what is happening. I've reinstalled the platformio-ide package, nothing changed. The module is there, for sure:

```
# ls -ld .atom/packages/platformio-ide/node_modules/fs-extra
drwxr-xr-x  9 sigxcpu  staff  306 Jul 29 11:23 .atom/packages/platformio-ide/node_modules/fs-extra
```

```
**Atom Version**: 1.8.0
**System**: Mac OS X 10.11.6
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package,v1.3.6

### Stack Trace

Failed to activate the platformio-ide package

At Cannot find module 'fs-extra'

Error: Cannot find module 'fs-extra'
    at Module._resolveFilename (module.js:338:15)
    at Function.Module._resolveFilename (/Applications/Atom.app/Contents/Resources/app.asar/src/module-cache.js:383:52)
    at Function.Module._load (module.js:289:25)
    at Module.require (module.js:366:17)
    at require (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:50:27)
    at module.exports (/Users/sigxcpu/.atom/packages/build-platformio/node_modules/promisify-node/index.js:157:15)
    at Object.<anonymous> (/Users/sigxcpu/.atom/packages/platformio-ide/lib/project-examples/command.js:27:13)
    at Module._compile (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/Applications/Atom.app/Contents/Resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:50:27)
    at Object.<anonymous> (/Users/sigxcpu/.atom/packages/platformio-ide/lib/main.js:37:66)
    at Module._compile (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:103:30)
    at Object.defineProperty.value [as .js] (/Applications/Atom.app/Contents/Resources/app.asar/src/compile-cache.js:208:21)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:313:12)
    at Module.require (module.js:366:17)
    at require (/Applications/Atom.app/Contents/Resources/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/Applications/Atom.app/Contents/Resources/app.asar/src/package.js:715:27)
    at Package.module.exports.Package.activateNow (/Applications/Atom.app/Contents/Resources/app.asar/src/package.js:173:16)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/package.js:156:32
    at Package.module.exports.Package.measure (/Applications/Atom.app/Contents/Resources/app.asar/src/package.js:92:15)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/package.js:149:26
    at Package.module.exports.Package.activate (/Applications/Atom.app/Contents/Resources/app.asar/src/package.js:146:34)
    at PackageManager.module.exports.PackageManager.activatePackage (/Applications/Atom.app/Contents/Resources/app.asar/src/package-manager.js:538:21)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/package-manager.js:519:29
    at Config.module.exports.Config.transactAsync (/Applications/Atom.app/Contents/Resources/app.asar/src/config.js:337:18)
    at PackageManager.module.exports.PackageManager.activatePackages (/Applications/Atom.app/Contents/Resources/app.asar/src/package-manager.js:514:19)
    at PackageManager.module.exports.PackageManager.activate (/Applications/Atom.app/Contents/Resources/app.asar/src/package-manager.js:497:46)
    at /Applications/Atom.app/Contents/Resources/app.asar/src/atom-environment.js:754:28
```

### Commands

```
```

### Config

```json
{
  "core": {
    "audioBeep": false,
    "disabledPackages": [
      "Tern",
      "go-plus",
      "term2",
      "autocomplete-elixir"
    ]
  }
}
```

### Installed Packages

```coffee
# User
atom-ternjs, v0.14.2
autocomplete-clang, v0.9.4
autocomplete-modules, v1.5.2
build, v0.65.0
build-platformio, v1.2.2
busy, v0.6.0
editorconfig, v1.4.1
language-elixir, v0.15.0
language-erlang, v3.0.0
language-hjson, v0.2.0
language-ini, v1.16.0
language-openscad, v0.4.3
language-rust, v0.4.6
language-scala-plus, v1.2.2
linter, v1.11.14
linter-elixirc, v1.1.10
linter-eslint, v7.2.4
linter-gcc, v0.6.15
linter-rust, v0.4.6
merge-conflicts, v1.4.4
pain-split, v1.4.0
platformio-ide, v1.3.6
platformio-ide-terminal, v2.1.1
tool-bar, v1.0.0

# Dev
No dev packages
```

sigxcpu76, 2016-07-29:
``` apm rebuild-module-cache``` doesn't solve the issue, either.

ivankravets, 2016-07-30:
See my answer here https://github.com/platformio/platformio-atom-ide/issues/95#issuecomment-234372130

Please reopen issue if you still have this problem.

closed 2016-07-30 by ivankravets