platformio/platformio-atom-ide #151: platformio not spawning pip during installation

auralmenace, 2016-08-12:
I have been unable to successfully install platformio on Atom 1.8.0 and platformio-IDE ver. 1.4.0 The platformIO about screen reports that the CLI version cannot be retrieved. 

I am running Atom on a MacbookPro running El Capitan 10.11.6. 

I have tried the solutions as described in issue #48 but I am _not installing behind a proxy_. Since Atom hangs and I can't cancel the process in order to get a traceback, I can't really provide much information.
[image]

When I restart, I get the second screen cap - PlatformIO IDE : Installing... : 
[image]
and then a prompt to reload.

I click on "Reload now" and as Atom is restarting, there is a red error block on the right side, but it goes by too fast for me to capture it.

I tried a manual installation using apm in bash: 
$ git clone https://github.com/platformio/platformio-atom-ide ~/.atom/packages/platformio-ide
$ cd ~/.atom/packages/platformio-ide
$ npm install

but got the error:
/usr/local/bin/npm: line 13: node: command not found

I also tried:
$ rm -rf ~/.atom/packages/platformio-ide/penv
$ sudo pip install virtualenv
$ virtualenv ~/.atom/packages/platformio-ide/penv

and got:
New python executable in /Volumes/Macintosh HD/Users/kevintillman/.atom/packages/platformio-ide/penv/bin/python2.7
Also creating executable in /Volumes/Macintosh HD/Users/kevintillman/.atom/packages/platformio-ide/penv/bin/python
Installing setuptools, pip, wheel...done.

However, 'platformio' still doesn't work.

[image]

I would really like to move out of the Arduino environment and platformIO has been recommended to me by a friend and I'd really like to be able to give it a spin - if I can get it to work.

Thanks.

ivankravets, 2016-08-15:
Try to remove `rm -rf ~/.atom/packages/platformio-ide/` and restart Atom.

auralmenace, 2016-08-15:
Thanks for the response. I get the blue
"Initializing PlatformIO IDE auto-installer. Please be patient and let the initialization to be completed." 
box

after a few seconds, I get the Java console and errors as mentioned above. In other words, no change.

ivankravets, 2016-08-15:
Could provide TeamViwer session for debug? Mail ID/PASS to me@ikravets.com

ivankravets, 2016-08-16:
The issue is linked with `virtualenv` and space in `$HOME` path. @auralmenace uses external storage/disk for `$HOME`. 

### Solution

Create virtual environment in the place where there is no space in the path and then symlink it to PlatformIO IDE. For example,
```
cd /tmp
virtualenv pioenv
ln -s /tmp/pioenv ~/.atom/packages/platformio-ide/penv
source pioenv/bin/activate
pip install platformio
```

Restart Atom

closed 2016-08-16 by ivankravets