platformio/platformio-atom-ide #152: Problem installing platform-atom-ide via Atom package manager

brainstorm, 2016-08-15:
I'm getting the following backtrace when installing platformio with my Atom 1.9.6:

[image]

I cannot see any virtualenvironment installed on `~/.atom/.../penv`, that subdirectory does not seem to exist:

[image]

Don't know what are the interactions between those Atom `penv` and my host system, but I use Miniconda on the host:

http://conda.pydata.org/miniconda.html

ivankravets, 2016-08-15:
Could you run these commands and restart Atom?

```
rm -rf ~/.atom/packages/platformio-ide/penv
virtualenv ~/.atom/packages/platformio-ide/penv
```

brainstorm, 2016-08-15:
That seems to be the issue, since I use [Anaconda](https://anaconda.org/) I have no "virtualenv" binary installed in my system:

```
$ rm -rf ~/.atom/packages/platformio-ide/penv
$ virtualenv ~/.atom/packages/platformio-ide/penv
-bash: virtualenv: command not found
```

So I went for:

```
$ conda create -n platformio python=2.7
$ conda-env list
# conda environments:
#
platformio               /Users/romanvg/.miniconda/envs/platformio
root                  *  /Users/romanvg/.miniconda
```

And then linking that venv to atom like this:

```
$ ln -sf /Users/romanvg/.miniconda/envs/platformio /Users/romanvg/.atom/packages/platformio-ide/penv
```

That did the trick :)

brainstorm, 2016-08-15:
But then another error just popped up:

[image]

:/

ivankravets, 2016-08-15:
PlatformIO IDE Installer downloads own `virtualenv` and make `penv`. Could try this test https://github.com/platformio/platformio-atom-ide/issues/146#issuecomment-235973537?

ivankravets, 2016-08-21:
@brainstorm do you still have problems?

brainstorm, 2016-08-21:
The IDE itself works well with my miniconda hack outlined above, so I'm happy now. Tried what you suggested and got this, hope you can fix the installation process for folks that already have Anaconda:

```
mba-3:virtualenv-14.0.6 romanvg$ python virtualenv.py /tmp/pio_penv
New python executable in /tmp/pio_penv/bin/python
Installing setuptools, pip, wheel...
  Complete output from command /tmp/pio_penv/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel:
  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/romanvg/tmp/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/__init__.py", line 13, in <module>
  File "/Users/romanvg/tmp/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/utils/__init__.py", line 18, in <module>
  File "/Users/romanvg/.miniconda/lib/python2.7/zipfile.py", line 6, in <module>
    import io
  File "/Users/romanvg/.miniconda/lib/python2.7/io.py", line 51, in <module>
    import _io
ImportError: dlopen(/tmp/pio_penv/lib/python2.7/lib-dynload/_io.so, 2): Symbol not found: __PyCodecInfo_GetIncrementalDecoder
  Referenced from: /tmp/pio_penv/lib/python2.7/lib-dynload/_io.so
  Expected in: flat namespace
 in /tmp/pio_penv/lib/python2.7/lib-dynload/_io.so
----------------------------------------
...Installing setuptools, pip, wheel...done.
Traceback (most recent call last):
  File "virtualenv.py", line 2304, in <module>
    main()
  File "virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "virtualenv.py", line 924, in create_environment
    download=download,
  File "virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /tmp/pio_penv/bin/python -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 1
```

brainstorm, 2016-08-21:
Actually, I just closed and opened again the editor and I'm not getting "Make sure this is a platformio project"... *sigh*.

ivankravets, 2017-02-21:
Could you try new PlatformIO IDE 2.0 Installer?

Menu: PlatformIO > Home > Try PlatformIO IDE 2.0 Preview.

Thanks!

ivankravets, 2017-02-21:
@brainstorm please try to remove `penv` folder and restart Atom. It should handle a built-in conda installation automatically

commit cecd5554 (2017-02-21): Use “conda” (if exists) to create virtual environment // Resolve #152 Resolve #174 Resolve #176 Resolve #209 Resolve #218
closed 2017-02-21 by ivankravets