platformio/platformio-atom-ide #155: Glitch // there is a platformio glitch for atom using the upload using programmer

djblackskull, 2016-08-23:
video reference: https://youtu.be/fYnJJlPV7QA
steps:
1: initializing a project
2: board: generic attiny85
3: glitch: when clicking upload using programmer from the bottom left, the popup menu that is displayed i've selected upload using programmer and what happens 2/3 times it doubles,triples,quadruples compilation to gcc.. stalling the compiler, and resulting in a for sure compile fault. it is so beyond annoying that i am leaving this message as a report.

having to use this method alone takes forever compared to different architectures/platforms since
this chip for whatever reason isn't fully supported. i've found myself using trinket3 as the env in platformio.
[env:trinket3]

platform = atmelavr
framework = arduino
board = trinket3

djblackskull, 2016-08-23:
video reference 2: https://youtu.be/vZt01OOINT8
info: this video shows what i mean a lot better. In this video I clicked upload using programmer only twice. The first click resulted in 4 compiles and the 2nd click sent one issue to compile yet resulted in two compiles.

ivankravets, 2016-10-01:
Could you upgrade Atom, PlatformIO-IDE package? Do you still have these issues?

ivankravets, 2016-10-17:
Please reopen if you still have problems.

closed 2016-10-17 by ivankravets