platformio/platformio-atom-ide #158: Separate platformio atom installation

ncolomer, 2016-08-28:
I'm using atom as my primary text editor for my every-day work, which has no relation to neither platformio nor electronics in general. When you install platformio package on an existing atom install, it just swaps the default UI, which is basically quite annoying since I only need platformio stuff when I do work on my personal projects.

It would be cool to have a separate and independent install of Platformio IDE.
Do you think this to be possible? Cause I didn't find anything close on the internet.

ivankravets, 2016-08-28:
Please go to Atom > Settings > Packages > Community Packages. Find here `platformio-ide` and click on "Disable". Restart Atom.

If you need enable PlatformIO IDE again, click "Enable" on plugin :)

ncolomer, 2016-08-29:
Already known the procedure but will be useful for future readers.

That's precisely what I wanted to avoid in this feature request (go to settings, find and disable the plugin + restart) since I already have intensive use of Atom in my everyday work and moreover don't wan't to mess my personal plugins / GUIs with platformio ones.

But I imagine (and understand) this to be far from current priorities ;)

s5kufabi, 2016-12-13:
This is thread is a bit older - but this might be helpful for other people:

I just had the same problem. My solution is to start atom with a different config folder just for PlatformIO.
Check this: https://github.com/atom/atom/pull/5385

thearduinoguy, 2017-01-15:
I have done the above but then the HOME tab goes missing so I have no way to create a new Python project. Nothing in FILE to create a new project for Python either.

s5kufabi, 2017-01-22:
Hey! For me this works:
Open command palette (Ctrl+Shift+P), type "platformio ide: home screen", enter

janstuemmel, 2017-02-28:
if youre on a unix like system, simply do an bash alias:

```sh
# the editor
alias platformio-ide='ATOM_HOME=$HOME/.platformio-ide atom $@'
# the package manager
alias piopm='ATOM_HOME=$HOME/.platformio-ide apm $@'
```

(add this to `~/.basrc` to make it permanent)

s5kufabi, 2017-02-28:
cool, thanks for sharing!

janstuemmel, 2017-02-28:
in addition, for my personal use i just created a `platformio-ide.desktop` file within `/usr/share/applications` with following contents:

```
[Desktop Entry]
Name=PlatformIO
GenericName=Text Editor
Exec=env ATOM_HOME=/home/[USERNAME]/.platformio-ide /opt/platformio-ide/atom %F
Icon=atom
Type=Application
StartupNotify=true
Categories=GNOME;GTK;Utility;TextEditor;Development;
MimeType=text/plain;
```
i rarely use PlatformIO, so i dont need the command line tools in my `PATH`, a `.desktop` file is enough for me

ivankravets, 2017-02-28:
I reopen this issue, it is useful for others. Also, I think we can gather all solutions and move them to docs.

janstuemmel, 2017-02-28:
a more complete solution would be to build a standalone version of atom editor with everything already installed. Could be done with a simple sh script:

```sh
#!/bin/bash

# installs platformIO IDE standalone on debian based machines

# add dirs
mkdir -p PlatformIO && cd PlatformIO &&

# downloads compiled atom by version or exits on 404
(wget https://github.com/atom/atom/releases/download/$1/atom-amd64.tar.gz ||
  (echo "No matching version found" && exit 1) ) &&

# extract
tar xzvf atom-amd64.tar.gz

# delete archive
rm *.tar.gz &&

# rename
mv atom-* atom &&

# add config folder
mkdir config &&

# add sh wrapper files
echo '#!/bin/bash' >> platformio-ide.sh &&
echo 'ATOM_HOME=$(dirname $(readlink -f $0))/config $(dirname $(readlink -f $0))/atom/atom $@' >> platformio-ide.sh &&
chmod +x platformio-ide.sh &&

echo '#!/bin/bash' >> piopm.sh &&
echo 'ATOM_HOME=$(dirname $(readlink -f $0))/config $(dirname $(readlink -f $0))/atom/resources/app/apm/bin/apm $@' >> piopm.sh &&
chmod +x piopm.sh &&

# install platformio
./piopm.sh install platformio-ide
```

there are a bunch of things you can do, like automate compiling for target architectures when a new version of atom arrives and for example then uploading binaries in the releases section of github...

punitsoni, 2017-05-09:
Do you have instructions to achieve this on Windows and Mac?

janstuemmel, 2017-05-09:
i do not have windows, but try downloading the `windows.zip` from [atom/releases](https://github.com/atom/atom/releases). Unpack it and start the Exe.

zeroby0, 2017-05-25:
Did anybody figure out a complete working method for macOS?

Thanks.

janstuemmel, 2017-05-25:
please read the suggestions above, try to compile the sources on your own, then open that application over the terminal and add the `ATOM_HOME` envar

tripflex, 2017-08-15:
I went ahead and created a custom application that will launch Atom using `~/.platformio-ide` as `ATOM_HOME` value, you can find it here:
https://github.com/tripflex/platformio-atom-osx

If anybody can take a look at the script file and let me know if they have any thoughts or suggestions on how to make it better?  

https://github.com/tripflex/platformio-atom-osx/blob/master/PlatformIO.sh

What would be the ultimate setup is if we could have two separate instances of Atom running, one for PlatformIO and one for Atom ... but unfortunately right now in my tests it seems if Atom is already open, for some reason calling a new instance does not respect the `ATOM_HOME` being set by my launcher application.

ivankravets, 2017-08-15:
@tripflex Thanks a lot for your work!

I think we can dynamically "Turn Off/On" PlatformIO IDE plugin depending on the opened projects. If we see "platformio.ini" in a project, we will enable the plugin.

Will that work for you?

janstuemmel, 2017-08-15:
> I think we can dynamically "Turn Off/On" PlatformIO IDE plugin depending on the opened projects. If we see "platformio.ini" in a project, we will enable the plugin.
> 
> Will that work for you?

bad idea when you have opened several projects on your workspace, thats why he created platform-atom-ide, so seperate applications for two seperate concerns

tripflex, 2017-08-15:
@ivankravets no problem, i'm sure as I get to understand how Atom works behind the scenes more when it comes to env vars and configuration I can start adding some more features, but wanted to at least get something simple out for now.

Regarding the `platformio.ini` file, I agree with @janstuemmel ... I use Atom as my daily driver for basic code/text editing (when don't need full blown jetbrains IDE), and love the PlatformIO integration ... but would rather have them separated.

My normal Atom install has a bunch of packages that I would never need in PlatformIO, as well as vice versa, numerous packages used in PlatformIO not used in my normal Atom usage ... and as such, I like having a completely separate setup and configuration for both 😄 

I first thought about integrating Atom build from source, but for now kept it simple and use the existing Atom install (so Atom upgrades are only required once) ... maybe later on can look at possibly build from source, but for now this workaround works nicely for me 👍

janstuemmel, 2017-08-15:
@tripflex consider linking this issue in your projects readme file