platformio/platformio-atom-ide #161: gcc linter can't find rtos.h error, but can build/upload sucess

mintisan, 2016-09-04:
I think it's not a functional bug, but may it's better to fix it.

Or maybe it's just not a bug. The description is below:

[image]

ps: 
0. I resetart my Atom, but the result is the same.
1. it's not same with [issue-24](https://github.com/platformio/platformio-atom-ide/issues/24)
2. u can tell me what information else u need to know, I will offer it.

version:
- macOS: 10.11.6
- Atom: 1.10.0
- PlatformIO IDE:1.4.0
- PlatformIO CLI:2.11.2
- Other
```
[env:nucleo_f401re]
platform = ststm32
framework = mbed
board = nucleo_f401re
```

ivankravets, 2016-09-04:
Please switch to development version with PlatformIO 3.0 (is going to be released next week). See http://docs.platformio.org/en/stable/installation.html#development-version

Then, Menu > Rebuild C/C++ Index.

Do you still have this issue?

mintisan, 2016-09-04:
@ivankravets Wow, so quickly.

The network in China is not so fast to use PlatformIO. Especially initialize a new platform(It cost me an hour to initialize my first mbed example). So I will tell u the result as soon as I `Reinstalling Initiated` completely.

ivankravets, 2016-09-04:
Thanks, keep me in touch. It's important for us because we are going to release PlatformIO 3.0. Need to be sure that you don't have this issue.

mintisan, 2016-09-04:
@ivankravets After `Reinstalling Initiated` completely. 
I restart Atom. The linter is just gone.

[image]

But When I `Menu > Rebuild C/C++ Index`, the Atom seems dead,,,

mintisan, 2016-09-04:
So I can only terminate it through `Force Quit`.

ivankravets, 2016-09-04:
PlatformIO IDE/Atom is wrapper around PlatformIO CLI. Looks that PlatformIO CLI upgrades from 2.0 to 3.0.

Yep, you can kill atom. Open it. Open PlatformIO IDE Terminal. Type something like `pio platform list`. What do you see?

mintisan, 2016-09-04:
[image]

[image]

mintisan, 2016-09-04:
I didn't install `nxplpc` manually ever...

ivankravets, 2016-09-04:
Does it work now?

mintisan, 2016-09-04:
It work well. The Rebuild is also OK.

The Atom is not dead, it gives a successful result for a while,,,,  ^_^!

Waiting for PlatformIO 3.0😄 

[image]

ivankravets, 2016-09-04:
PlatformIO 3.0 is totally rewritten PlatformIO. We will receive a lot of new features and improvements. Please follow us on Twitter https://twitter.com/PlatformIO_Org to be notified about announcements.

closed 2016-09-04 by ivankravets