platformio/platformio-atom-ide #162: The first use and immediately bug?

profra, 2016-09-05:
I read and heard only praises for Platformio and so I decided to try it out... (OS Linux Mint). I tried to translate simple program Blink in two environments (two different project folder)...

1/
[env:pro8MHzatmega328]
platform = atmelavr
framework = arduino
board = pro8MHzatmega328 .... without any problem

2/
[env:d1_mini]
platform = espressif
framework = arduino
board = d1_mini ... with the the following errors...

I have received the same errors in Atom IDE and also in CLI.
I am not sure if it is bug or not... but I don't know the way how to solve it.
Any help?
 
```
The next files/directories will be created in /home/fp/opt/aa_myprojects/esp_01
platformio.ini - Project Configuration File. |-> PLEASE EDIT ME <-|
src - Put your source files here
lib - Put here project specific (private) libraries
Do you want to continue? [y/N]: y

Project has been successfully initialized!
Useful commands:
`platformio run` - process/build project from the current directory
`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information
fp@fp-VC62B ~/opt/aa_myprojects/esp_01 $ pio run
[Mon Sep  5 03:24:49 2016] Processing d1_mini (platform: espressif, board: d1_mini, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
The platform 'espressif' has not been installed yet. Would you like to install it now? [y/N]: y
Installing tool-scons package:
Already installed
Installing toolchain-xtensa package:
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Installing tool-esptool package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Installing framework-arduinoespressif package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Installing ldscripts package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
IOError: [Errno 2] No such file or directory: '/home/fp/.platformio/packages/framework-arduinoespressif/version.txt':
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/main.py", line 159:
env.SConscript("$BUILD_SCRIPT")
File "/home/fp/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/fp/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/scripts/espressif.py", line 308:
target_elf = env.BuildProgram()
File "/home/fp/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 62:
f.lower().strip() for f in env.get("FRAMEWORK", "").split(",")])
File "/home/fp/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 248:
join("$PIOBUILDER_DIR", "scripts", "frameworks", "%s.py" % f)))
File "/home/fp/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 614:
return method(*args, **kw)
File "/home/fp/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/fp/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/scripts/frameworks/arduino.py", line 183:
"version.txt")).read().replace(".", "").strip())
================================================================ [ ERROR ] Took 15.67 seconds ================================================================
fp@fp-VC62B ~/opt/aa_myprojects/esp_01 $ 
```

ivankravets, 2016-09-05:
Please sorry, this is my big mistake. I broke Arduino framework for ESP8266 package yesterday. 

Please run `Menu: PlatformIO: > Update packages`. Does it work now?

profra, 2016-09-05:
Thank you for fast response ... we had both really bad day yesterday :-) ... I did pio update in CLI... and Yay.... 'pio run' + 'pio run --target upload' and it is OK ... LED is blinking on both boards!!!
I am fresh user of platformio, I have no experience with it .... but I would have one question more... I am used to use UltraEdit Professional Editor "www.ultraedit.com" for years... Is there any possibility (way) to port IDE also into this editor (or UEStudio)? UE is used by over 2 milions of developers in the whole world...
Many thanks in advance for your reply.

ivankravets, 2016-09-05:
Please make feature request here https://github.com/platformio/platformio/issues

Happy Coding with PlatformIO!

closed 2016-09-05 by ivankravets