platformio/platformio-atom-ide #163: PLATFORMIO_LIB_DIR env variable not honored

bcelary, 2016-09-06:
Looks like the PLATFORMIO_LIB_DIR variable is no longer used by the atom ide. I've tried to disable using of built-in PlatformIO but does not change anything. I am able to compile from the system terminal (Linux).

ivankravets, 2016-09-07:
Are you sure that this environment is set in the same shell where Atom is run?

bcelary, 2016-09-08:
OK. Not an issue. It was working because I've started atom from console where the bashrc is properly configured. Then started from the launcher.

But it would be good to have an option for these setting in the plugin. Is there a way to configure it from atom somehow?

Thanks!

ivankravets, 2016-09-08:
Why do you need that? Use PlatformIO 3.0 new feature name `lib_extra_dirs`. See http://docs.platformio.org/en/stable/projectconf.html#lib-extra-dirs

bcelary, 2016-09-08:
OK. Looks like I might be doing something wrong here if that's not a common case.

So I'd like to keep my libs in e.g. ~/Arduino/lib and the projects in ~/Arduino/src

What's the recommended way of having that?

ivankravets, 2016-09-08:
You can use 
```ini
[env:...]
lib_extra_dirs = ~/Arduino/lib
```

and share this folder between project. I'm not sure that relative path works. If it doesn't work, this is a bug. Try full path and open issue here https://github.com/platformio/platformio/issues

bcelary, 2016-09-08:
You mean to add it to the platformio.ini for the project? But then I would need to add it every time to the project which I'd like to avoid.

ivankravets, 2016-09-08:
If you don't want to touch `platformio.ini`, there are options:

1. http://docs.platformio.org/en/stable/projectconf.html#lib-dir
2. Make symlink to `~/.platformio/lib` (this is global internal storage)

However, I don't like these 2 ways. `lib_extra_dirs` for a project is a right way. Maybe, need to allow configure it via environment variable too.

closed 2016-09-09 by bcelary