platformio/platformio-atom-ide #166: Internal error indexedDB.open

nooitaf, 2016-09-13:
[image]

I get these errors and screen at every atom startup (even though i unchecked "show on startup") and everytime it tries to update something (every 30m or so) i get those errors again. For the rest it works fine and i'm very happy.

ivankravets, 2016-09-13:
Please open a developer console (click "View -> Developer -> Toggle Developer Tools") paste the string `indexedDB.deleteDatabase('platformio-ide')` and restart the editor.

nooitaf, 2016-09-13:
I assume you mean paste it in the console, did that (twice), didn't fix it. 
Tried to update to 1.5.0, also didn't work.

[image]

ivankravets, 2016-09-13:
What is your OS?

nooitaf, 2016-09-13:
Linux Mint Sarah 18 
4.4.0-28-generic #47-Ubuntu x86_64 x86_64 x86_64 GNU/Linux

ivankravets, 2016-09-13:
Could you try to remove DB again https://github.com/platformio/platformio-atom-ide/issues/166#issuecomment-246757091 and then `rm -rf ~/.atom`. Restart IDE.

Very strange error.

nooitaf, 2016-09-13:
Well .. `rm -rf ~/.atom` deleted all my settings and packages, so i tried again. Same thing..
It's Atom version `1.8.0` btw

[image]

[image]

So i tried reinstalling atom..
`sudo apt-get remove atom`
downloaded & installed `atom.deb` `v1.10.2`
installed `platformio-ide` package

[image]

[image]

ivankravets, 2016-09-13:
Where is Atom application folder in your OS? See hint for macOS https://github.com/atom/atom/issues/11254#issuecomment-200355845

orgkhnargh, 2016-09-13:
Neither reinstalling atom nor removing the `~/.atom` directory would not clear the indexeddb files, because they are stored under the `~/.config/Atom/IndexedDB/` directory. @nooitaf, you'll want to remove this one to make the error disappear (make a backup first just in case).

Do you have enough space on the hard drive? Someone [here](https://bugs.chromium.org/p/chromium/issues/detail?id=279106) said: 

> when I try to add more data than the available space, I get no error, success is received instead, disk gets full, and I don't get any error on page refresh, even though db is loaded again at page loading. Only when Chrome is closed and reopened and try to open the db, I get the above error.

nooitaf, 2016-09-13:
@orgkhnargh That was it! Awesome \o/

I started Atom as root sometime back. So i had to re-own some of the index files...

`sudo chown -R username:username ~/.config/Atom` 

...fixed it. But this error didn't show before so i didn't notice it and thought it was related :O 

Thanks guys!

closed 2016-09-13 by nooitaf