platformio/platformio-atom-ide #169: Uncaught TypeError: Cannot read property 'startsWith' of undefined

stelgenhof, 2016-09-24:
Opened Atom again today and received below error report. Apparently it is related to the new Atom 1.10.2 version. Upgraded all packages (PlatformIO and Atom), but no luck and error keeps on coming. 

**Atom Version**: 1.10.2
**Electron Version**: 0.37.8
**System**: Mac OS X 10.11.6
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package, v1.6.0

### Stack Trace

Uncaught TypeError: Cannot read property 'startsWith' of undefined

```
At /Users/sachatelgenhofoudekoehorst/.atom/packages/platformio-ide/lib/build-provider.js:107

TypeError: Cannot read property 'startsWith' of undefined
    at /Users/sachatelgenhofoudekoehorst/.atom/packages/platformio-ide/lib/build-provider.js:107:25
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:404:3)
```

### Commands

```
```

### Config

```json
{
  "core": {
    "ignoredNames": [
      ".git",
      ".hg",
      ".svn",
      ".DS_Store",
      "._*",
      "Thumbs.db",
      ".pioenvs",
      ".piolibdeps",
      ".clang_complete",
      ".gcc-flags.json"
    ]
  },
  "platformio-ide": {
    "showHomeScreen": false,
    "showPlatformIOFiles": true
  }
}
```

### Installed Packages

```coffee
# User
atom-beautify, v0.29.12 (inactive)
autocomplete-clang, v0.10.0 (active)
build, v0.65.0 (active)
busy, v0.7.0 (active)
language-ini, v1.16.0 (active)
linter, v1.11.16 (active)
linter-gcc, v0.6.15 (active)
platformio-ide, v1.6.0 (active)
platformio-ide-terminal, v2.2.0 (active)
tool-bar, v1.0.1 (active)
atom-dark-syntax, v0.27.0 (inactive)
atom-dark-ui, v0.52.0 (inactive)
atom-light-syntax, v0.28.0 (inactive)
atom-light-ui, v0.44.0 (inactive)
base16-tomorrow-dark-theme, v1.1.0 (inactive)
base16-tomorrow-light-theme, v1.1.1 (inactive)
one-dark-ui, v1.5.0 (active)
one-light-ui, v1.5.0 (inactive)
one-dark-syntax, v1.3.0 (active)
one-light-syntax, v1.3.0 (inactive)
solarized-dark-syntax, v1.0.2 (inactive)
solarized-light-syntax, v1.0.2 (inactive)
about, v1.5.3 (active)
archive-view, v0.61.1 (active)
autocomplete-atom-api, v0.10.0 (active)
autocomplete-css, v0.11.2 (active)
autocomplete-html, v0.7.2 (active)
autocomplete-plus, v2.31.1 (active)
autocomplete-snippets, v1.11.0 (active)
autoflow, v0.27.0 (inactive)
autosave, v0.23.1 (active)
background-tips, v0.26.1 (active)
bookmarks, v0.41.0 (active)
bracket-matcher, v0.82.1 (active)
command-palette, v0.38.0 (inactive)
deprecation-cop, v0.54.1 (active)
dev-live-reload, v0.47.0 (active)
encoding-selector, v0.22.0 (active)
exception-reporting, v0.39.0 (active)
find-and-replace, v0.201.0 (inactive)
fuzzy-finder, v1.3.0 (active)
git-diff, v1.1.0 (active)
go-to-line, v0.31.0 (inactive)
grammar-selector, v0.48.1 (active)
image-view, v0.58.2 (active)
incompatible-packages, v0.26.1 (active)
keybinding-resolver, v0.35.0 (active)
line-ending-selector, v0.5.0 (active)
link, v0.31.1 (inactive)
markdown-preview, v0.158.0 (active)
metrics, v0.53.1 (active)
notifications, v0.65.0 (active)
open-on-github, v1.2.0 (inactive)
package-generator, v1.0.0 (inactive)
settings-view, v0.242.2 (active)
snippets, v1.0.2 (active)
spell-check, v0.67.1 (active)
status-bar, v1.4.1 (active)
styleguide, v0.47.0 (active)
symbols-view, v0.113.0 (inactive)
tabs, v0.100.2 (active)
timecop, v0.33.2 (active)
tree-view, v0.208.2 (active)
update-package-dependencies, v0.10.0 (active)
welcome, v0.34.0 (active)
whitespace, v0.33.0 (active)
wrap-guide, v0.38.1 (active)
language-c, v0.52.1 (active)
language-clojure, v0.21.0 (active)
language-coffee-script, v0.47.2 (active)
language-csharp, v0.12.1 (active)
language-css, v0.37.1 (active)
language-gfm, v0.88.0 (active)
language-git, v0.15.0 (active)
language-go, v0.42.1 (active)
language-html, v0.45.1 (active)
language-hyperlink, v0.16.0 (active)
language-java, v0.23.0 (active)
language-javascript, v0.119.0 (active)
language-json, v0.18.2 (active)
language-less, v0.29.5 (active)
language-make, v0.22.2 (active)
language-mustache, v0.13.0 (active)
language-objective-c, v0.15.1 (active)
language-perl, v0.35.0 (active)
language-php, v0.37.2 (active)
language-property-list, v0.8.0 (active)
language-python, v0.45.0 (active)
language-ruby, v0.69.0 (active)
language-ruby-on-rails, v0.25.0 (active)
language-sass, v0.56.0 (active)
language-shellscript, v0.22.4 (active)
language-source, v0.9.0 (active)
language-sql, v0.23.0 (active)
language-text, v0.7.1 (active)
language-todo, v0.28.0 (active)
language-toml, v0.18.0 (active)
language-xml, v0.34.9 (active)
language-yaml, v0.26.0 (active)

# Dev
No dev packages
```

ivankravets, 2016-09-24:
Hmm... Could you close IDE and `rm -rf ~/.atom/packages/platformio-ide`? Start IDE.

stelgenhof, 2016-09-24:
I found out the culprit. It was the `platformio.ini` file of my project causing this issue. Once I reverted the changes made I had no issue. What exactly the issue was, I will try to find out :)

ivankravets, 2016-09-24:
That is very interesting. Looks like broken INI file. Could you share it here?

stelgenhof, 2016-09-24:
Sure! I figured it out: the error message appears when an env: section doesn't contain the `platform` setting.

Simple example (before), that works without any issue:

```
[env:nodemcuv2]
board = nodemcuv2
platform = espressif8266
framework = arduino
```

Example that generates the error:

```
[platformio]
env_default = nodemcuv2

[common]
platform = espressif8266
framework = arduino

[env:nodemcuv2]
include = common
board = nodemcuv2
```

I was trying to create separate environments for different platforms including a common setting. I guess `platform` is a required variable :)

ivankravets, 2016-09-24:
What does mean "include = common"? Where did you find it? 

See docs http://docs.platformio.org/en/stable/projectconf.html#dynamic-variables

stelgenhof, 2016-09-24:
I found it here: https://bitbucket.org/xoseperez/rentalito-esp8266/src/d83d03e096e0cc18829163bacbb568ae94c5c4bc/code/platformio.ini?at=master&fileviewer=file-view-default

Just was trying things out :)

ivankravets, 2016-09-24:
I got you. This author uses an own fork of PlatformIO with own modifications to it. Also, he has made a PR to our repository. See my comment here https://github.com/platformio/platformio/pull/790

stelgenhof, 2016-09-24:
Ah I see. Sorry about the confusion. Thanks for closing the issue.

ivankravets, 2016-09-24:
Please use official dynamic variables documented here http://docs.platformio.org/en/stable/projectconf.html#dynamic-variables

joscha, 2017-03-08:
This can actually also happen with an incorrect config:

```
[env:sodaq_one]
platform = atmelsam
framework = arduino
board = sodaq_one

[env:dev]

```

commit 86168280 (2017-03-24): Refactor Build Provider // Resolve #169
commit fe5f4392 (2018-03-03): Merge branch 'develop'

* develop: (127 commits)
  Prepare 2.0.0-beta.1 release
  Update Python Installation link
  Update PIO Core Packages and depend on a new PIO Core >=3.3
  Cosmetic update
  Fix issue with example toggling
  “Library Built-in” description; cosmetic improvements // Resolve #241
  Bump version to 2.0.0-alpha.14
  Allow to use develop version of IDE; new settings in a package; automatically install IDE updates
  Add command/hotkey for PIO Core `program` target // Resolve #154
  Fix multiple Atom commands for the same build target
  Disable conflicted packages
  Bump version  [... cut]
closed 2017-03-24 by ivankravets