platformio/platformio-atom-ide #170: can't deinstall platformio

tennismichel, 2016-09-29:
When I updated platformio to the latest version, it started to disappear in Atom. It still says that platformio is installed as a package, but the icon of the package is kind of broken and I can't deinstall it. I tried to reinstall the whole thing using these commands:
rm -rf ~/.atom
rm -rf /usr/local/bin/atom
rm -rf /usr/local/bin/apm
rm -rf /Applications/Atom.app
rm -rf ~/Library/Preferences/com.github.atom.plist
rm -rf ~/Library/Application Support/com.github.atom.ShipIt
rm -rf ~/Library/Application Support/Atom
rm -rf ~/Library/Saved Application State/com.github.atom.savedState
rm -rf ~/Library/Caches/com.github.atom
rm -rf ~/Library/Caches/Atom

But it didn't make a difference. You guys have any idea what I could do? I tried to uninstall it using the terminal, but at first the pip commands weren't available. When I figured that out, I was able to get a successful installation and uninstallation via the terminal (according to the terminal) but in Atom itself nothing changed. Still saying that the platformio package is installed, but it's showing a weird icon...

Thank you very much for your support. I really appreciate it. 

Michi

closed 2016-09-29 by tennismichel

ivankravets, 2016-09-29:
Did you resolve it?

tennismichel, 2016-09-29:
yeah. I had to use all the commands from above with sudo. I realized some parts of the Uninstallation didn't work without, so that helped to deinstall/delete everything and then I was able to install a working platformio.