platformio/platformio-atom-ide #171: Failed to initialize PlatformIO Project

Lord-Frost, 2016-10-01:
I was able to install PlatformIO IDE using Atom but when initializing project, I'm getting this error:
"Cannot install 'toolchain-atmelavr' with version requirements '~1.40801.0' for your system 'windows_amd64'"

I'm using Atom 1.10.2 running on Windows 10 Home Single Language
PlatformIO IDE: 1.6.0 CLI: 3.1.0

[image]
.

ivankravets, 2016-10-01:
Please open `Menu: PlatformIO > Terminal > Toggle` and type this command
```
pio platform install atmelavr
```

Could you share here an output?

Lord-Frost, 2016-10-02:
This is what I get as output:
`PS C:\Users\aravi\.atom\packages\platformio-ide> pio platform install atmelavr
PlatformManager: Installing atmelavr
atmelavr @ 1.0.2 is already installed
PackageManager: Installing toolchain-atmelavr @ ~1.40801.0
Downloading...
Unpacking...
Warning! Package Mirror: [Error 5] Access is denied
Looking for the another mirror...
Downloading...
Unpacking...
Warning! Package Mirror: [Error 5] Access is denied
Looking for the another mirror...
Error: Can not install 'toolchain-atmelavr' with version requirements '~1.40801.0' for your system 'windows_amd64'`

[image]

Lord-Frost, 2016-10-04:
Hello?

ivankravets, 2016-10-04:
1. Close IDE
2. Please remove manually `C:\Users\aravi\.platformio` folder
3. Open IDE
4. Open the same Terminal within IDE
5. Run `pio platform install atmelavr`

Lord-Frost, 2016-10-04:
Thank you! The toolchain has downloaded. I haven't tested it out on a board but will update you with the status once I do.

Lord-Frost, 2016-10-05:
Tested on Arduino Uno. Works like a charm! Issue can be closed.

closed 2016-10-08 by ivankravets

dontsovcmc, 2016-10-20:
I got this error on Windows 10 when install Platformio CLI from Python script:

1) python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
2) platformio run --upload-port COM3

Error: "Cannot install 'toolchain-atmelavr' with version requirements '~1.40801.0' for your system 'windows_amd64'"

Try:
3) pio platform install atmelavr
also Error: "Cannot install 'toolchain-atmelavr' with version requirements '~1.40801.0' for your system 'windows_amd64'"

I don't have "C:\Users\aravi\.platformio" to remove :(

ivankravets, 2016-10-20:
You don't need to install PlatformIO CLI separately. It's shipped with PlatformIO IDE. Please open PlatformIO IDE Terminal and use it.

dontsovcmc, 2016-10-20:
I only need command line interface :)
I don't understand advantages of Atom for small projects :((((

ivankravets, 2016-10-20:
Why do you post in `platformio-atom-ide` repository? :)

P.S: Open Windows CMD and run `pip install platformio`.

dontsovcmc, 2016-10-20:
Sorry, I found the same bug by google here :)

ivankravets, 2016-10-20:
@dontsovcmc does it work now?

dontsovcmc, 2016-10-20:
No:

```
$ platformio run
[10/20/16 14:40:46] Processing uno (platform: atmelavr, upload_port: COM16, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-atmelavr @ ~1.40801.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Warning! Package Mirror: [Error 5]
Looking for the another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Warning! Package Mirror: [Error 5]
Looking for the another mirror...
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Warning! Package Mirror: [Error 5]
Looking for the another mirror...
Error: Can not install 'toolchain-atmelavr' with version requirements '~1.40801.0' for your system 'windows_amd64'
```

ivankravets, 2016-10-20:
http://stackoverflow.com/questions/28528020/why-am-i-getting-windowserror-error-5-access-is-denied

1. Don't use Administrative privileges to work with PIO.
2. Remove `C:\Users\<my user>\.platformio` if exists

dontsovcmc, 2016-10-20:
I run command line with Administrative provileges and execute:
$ platformio run
toolchain-atmelavr and others installed successfully!

So, it works!

ivankravets, 2016-10-20:
@dontsovcmc Happy Coding with PlatformIO! 😊

dontsovcmc, 2016-10-20:
Thanks, Ivan! :))

fredator, 2016-11-20:
I don't understand this. You say to NOT use platformio with admin privileges so he runs as admin and it works?? What if you're just using the ide in atom? I get the same error and don't know how to fix it.

dontsovcmc, 2016-11-20:
It is incorrect to install platformio (not atom, only command line tool) from python script.
You should use pip: pip install platformio

fredator, 2016-11-20:
I've done this but now the package doesn't show up at all. How do I build as a platformio project now?

dontsovcmc, 2016-11-20:
Type "platformio" in command line. Do you see correct info?
If yes, you should do platformio.ini file in you project folder :). See help site.

fredator, 2016-11-20:
Still same error, running as administrator or not as administrator.. sigh..

frjojoToulouse, 2017-03-05:
Hello
I do not speak English
configuration:
Processor intel I3
Window 10 64
Python version 2.7.13
Atom 1.14.4
IDE 1.7.3 | Core 3.2.1
I want to install pio for a project 'espressif2866' but I have a bug at startup:

Error: Can not install 'espressif8266' with version requirements '*' for your system 'windows_x86'

[image]

thank you

ivankravets, 2017-03-05:
Please switch to PlatformIO IDE 2.0 and try install development platform from `PIO Home > Platforms > Install`

How to switch to PIO IDE 2.0 from 1.7 version? Menu: PlatformIO > Home > "Switch to PIO IDE 2.0"

frjojoToulouse, 2017-03-05:
Switch to PIO IDE 2.0 Ok

Same error

PlatformIO: Failed to initialize PlatformIO project!
Error: Can not install 'espressif8266' with version requirements '*' for your system 'windows_x86'

ivankravets, 2017-03-05:
@Lord-Frost Home  > Platforms > ESP8266 > Install

What do you see?

frjojoToulouse, 2017-03-05:
Platform command failed: platform install espressif8266@1.3.0
Error: Error: Can not install 'espressif8266' with version requirements '1.3.0' for your system 'windows_x86'
Report a problem
[image]

frjojoToulouse, 2017-03-05:
for ESP8266

Library command failed: lib --global install 127
Error: Error: Traceback (most recent call last):
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\__main__.py", line 106, in main
    cli(None, None, None)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\commands\lib.py", line 97, in lib_install
    lm.install(library, silent=silent, interactive=interactive)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\lib.py", line 276, in install
    if not _url else name, _requirements, silent, trigger_event)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\package.py", line 565, in install
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\lib.py", line 255, in _install_from_piorepo
    if app.get_setting("enable_ssl") else dl_data['url'], requirements)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\package.py", line 324, in _install_from_url
    pkg_dir = self._install_from_tmp_dir(pkg_dir, requirements)
  File "c:\users\jojo\.atom\packages\platformio-ide\penv\lib\site-packages\platformio\managers\package.py", line 381, in _install_from_tmp_dir
    os.rename(tmp_dir, pkg_dir)
WindowsError: [Error 5] Acc�s refus

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

Report a problem

Run    pip install -U platformio   ------->Same error

frjojoToulouse, 2017-03-06:
Hello
After testing, platformio does not work with python 3.
Python 2.7 and 3 had been installed on the pc
I have uninstalled, shut down antivirus, and then reinstalled Python2.7 and platformio 2
Then I installed the package 'espressit8266'
Everything is OK
Thank you

frjojoToulouse, 2017-03-06:
Hello
After testing, platformio does not work with python 3.
Python 2.7 and 3 had been installed on the pc
I have uninstalled, shut down antivirus, and then reinstalled Python2.7 and
platformio
Then I installed the package 'espressit8266'
Everything is OK
Thank you

ivankravets, 2017-03-06:
@frjojoToulouse There is no problem that you have multiple Pythons in a system if they are specified in PATH. PlatformIO IDE Installer will find correct Python automatically.

In your case, the problem was linked with Antivirus. See error message above `WindowsError: [Error 5] Acc�s refus`. It means, "Access refused", PlatformIO Core can't download and install any toolchains, frameworks.

What is your Antivirus?

frjojoToulouse, 2017-03-06:
I had 2 antivirus: mcafee and avast, I just bought the pc and during the installation it added me Avast. So I deleted avast and inhibited McAfee for installation.

Now I have another problem it does not build.

"PlatformIO: Failed to run a command: Build
Please make sure that "build" package is installed and activated."

#include <Arduino.h>
void setup() {
  Serial.begin(9600);
}
void loop() {
  delay(1000);
  Serial.println("Hello World!");
}

build0.67.0  and build-xcodebuild Installed
thank you

studuino, 2017-03-30:
I have platformio IDE problems as follows:
1. PlatformIO Menu options like "PlatformIO Home", "Project Examples", "Initialize or Update PlatformIO Projects", etc. are not responding/ working at all.

2. After atom start-up, there is a pop at the right upper corner to restart Atom since PlatformIO IDE has been successfully installed.

I've tried deleting ".platformio" and restarted "Atom" but no avail.
I've also upgraded PlatformIO core and installed packages.

would you please advice?

PS. I can build, upload using F7 (Run other target)

ivankravets, 2017-03-30:
@studuino do you use the latest PIO IDE 2.0?

studuino, 2017-03-30:
@ivankravets Yes, I did.
ATOM version is 1.15.0 ia32 
platformio-ide 2.0.0-alpha.9
platformio-ide-terminal 2.5.0

ivankravets, 2017-03-30:
> 2.0.0-alpha.9

We had a problem with auto-updating for the first alphas. Currently, 2.0 Beta 2 is available.

Please 

1. Atom > Settings > Packages > serch for `platformio-ide` and uninstall `2.0.0-alpha.9`
2. Atom > Settings > Install > Install the latest beta

studuino, 2017-03-30:
@ivankravets Thanks, that solved the problem.