platformio/platformio-atom-ide #172: issue with path

drepou, 2016-10-21:
failed to compile if directory contain a blank caracter :

10/21/16 16:36:27] Processing nodemcu (targets: upload, platform: espressif8266, board: nodemcuv2, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Collected 24 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ESP8266WiFi> v1.0 (D:\WINDOWS SSD\drepou\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi)
|-- <Octosnake> (D:\bricolage et electronique\electronique\robots\miniKame-develop\miniKame-develop\code\arduino\lib\Octosnake)
|   |-- <Servo(esp8266)> v1.0.2 (D:\WINDOWS SSD\drepou\.platformio\packages\framework-arduinoespressif8266\libraries\Servo)
|-- <Servo(esp8266)> v1.0.2 (D:\WINDOWS SSD\drepou\.platformio\packages\framework-arduinoespressif8266\libraries\Servo)
"esptool" -eo D:\WINDOWS SSD\drepou\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf -bo .pioenvs\nodemcu\firmware.bin -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo .pioenvs\nodemcu\firm
ware.elf -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
error: can't open ELF file D:\WINDOWS
*** [.pioenvs\nodemcu\firmware.bin] Error -1

ivankravets, 2016-10-22:
Please move this issue to https://github.com/platformio/platform-espressif8266/issues

ivankravets, 2016-12-01:
Fix, please run `pio update`

closed 2016-12-01 by ivankravets