platformio/platformio-atom-ide #173: "TypeError: Cannot read property 'dataset' of null" when opening files

kleerkoat, 2016-10-21:
getting the following error when opening any file type in Atom.

```
/Users/rgilmore/.atom/packages/platformio-ide/lib/maintenance.js:221
Hide Stack Trace
TypeError: Cannot read property 'dataset' of null
    at /Users/rgilmore/.atom/packages/platformio-ide/lib/maintenance.js:221:39
    at Array.forEach (native)
    at Object.highlightActiveProject (/Users/rgilmore/.atom/packages/platformio-ide/lib/maintenance.js:213:8)
    at doHighlight (/Users/rgilmore/.atom/packages/platformio-ide/lib/main.js:247:43)
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:25:14)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app.asar/node_modules/event-kit/lib/emitter.js:129:28)
    at stoppedChangingActivePaneItemCallback (/Applications/Atom.app/Contents/Resources/app.asar/src/pane-container.js:350:36)
```

ivankravets, 2016-10-22:
Which version of Atom and PlatformIO IDE do you use? Click on `Menu: PlatformIO > Home`.

ivankravets, 2017-02-21:
1. Update Atom to the latest version ( http://atom.io )
2. Update PIO IDE to the latest version.

Please reopen if you still have problems.

closed 2017-02-21 by ivankravets