platformio/platformio-atom-ide #174: Error installing IDE when python is installed in custom path

skuhtic, 2016-10-22:
Installation is unable to find python when it is install in folder like c:\python\python27. There is path in PATH variable and python is working from command line. Installation is suggesting checking the PATH variable, installing python or aborting installation of Platform IDE. 

I installed python in default folder and it worked. It should work with other paths used or there should be option for manual locating the installation python path.

It is easy to fix situation, but i guess it's easy to fix to work as suggested (using PATH variable).

ivankravets, 2016-10-22:
If you have it in a PATH - it should work! See source code [here](https://github.com/platformio/platformio-atom-ide/blob/develop/lib/utils.js#L39).

As you can see, we check all possible locations. Could you try to specify both settings: `Path` in system section and `PATH` in user's section. I'm not Windows user but I saw these options when edit system environment variables.

skuhtic, 2016-10-22:
I did try all combinations with user and system PATH variables. with and without trailing backslashes and renamed python folder to 'c:\python' but it does not work. When renamed to 'c:\python27' works even if variable are still set to 'c:\python' (and then command line does not work). Hope that helps.

ivankravets, 2016-10-22:
Please open PlatformIO IDE Terminal and run 
```
echo %PATH%
echo %Path%
```

What do you see?

skuhtic, 2016-10-22:
Same variables with python the first one.

ivankravets, 2016-10-22:
Thanks, I'll take a look on it later.

skuhtic, 2016-10-22:
C:\>echo %Path%
C:\Python\;C:\Python\Scripts;C:\ProgramData...
I got it working and it's easy to fix... I don't know how often python is installed in custom path so it's maybe minor issue. Thnx for all the effort...

skuhtic, 2016-10-22:
I didn't test in IDE terminal but in cmd and now I have different result. Sorry, my mistake. So to not mislead you, this is the output:

Windows PowerShell
Copyright (C) 2016 Microsoft Corporation. All rights reserved.

PS C:\Users\xxx> cd /
PS C:\> echo %PATH%
%PATH%
PS C:\> echo %Path%
%Path%
PS C:\>

PS Keyboard is english and not the one I'm using in windows...

ivankravets, 2016-10-22:
Ask google how to print current env var using PowerShell

skuhtic, 2016-10-22:
Allredy did ;)

PS C:\> echo $env:PATH
C:\Users\Sinisha\.atom\packages\platformio-ide\penv\Scripts;C:\Python27\;C:\Python27\Scripts;C:\ProgramData

Link on the issue:
http://stackoverflow.com/questions/714877/setting-windows-powershell-path-variable

ivankravets, 2016-10-22:
Do you see here `C:\Python\;C:\Python\Scripts`? I don't.

ivankravets, 2016-10-22:
Atom doesn't see your custom environment PATHs. Looks like you set them incorrect.

skuhtic, 2016-10-22:
There is: C:\Python27\;C:\Python27\Scripts i changed it to original...
After changing back to 'c:\python':

PS C:\> echo $env:Path
C:\Users\Sinisha\.atom\packages\platformio-ide\penv\Scripts;**C:\Python\;C:\Python\Scripts**;C:\ProgramData

ivankravets, 2016-10-22:
Cool, I've jsut changed this issue to "bug". I'll fix it later.

kvij, 2016-11-07:
I have been trying to set it up for a few hours now. I did have a pip problem because strawberry perl also has a pip command in PATH. When I use cmd or PS or even Komodo IDE and type "python" I have python 2.7.11 running (chocolatey install) so PATH is setup correctly. I also need python 3 so I am not going to uninstall that.

ashthespy, 2017-01-26:
I seem to have a peculiar issue: 
OS: Win 8.1
I have `pyhton 2.7.12` installed via `miniconda` and is registered and available at the cmd prompt and PS. 
however, Atom and platformIO don't seem to be able to find python. 
Any ideas? Works fine if I install Python for Windows directly, but I would like to avoid two installs of Python on my system.

ivankravets, 2017-02-21:
Could you try new PlatformIO IDE 2.0 Installer?

Menu: PlatformIO > Home > Try PlatformIO IDE 2.0 Preview.

Thanks!

ivankravets, 2017-02-21:
@ashthespy please try to remove `penv` folder and restart Atom. It should handle a built-in conda installation automatically

commit cecd5554 (2017-02-21): Use “conda” (if exists) to create virtual environment // Resolve #152 Resolve #174 Resolve #176 Resolve #209 Resolve #218
closed 2017-02-21 by ivankravets