platformio/platformio-atom-ide #176: installation problems a little help pls

smunozeyquem, 2016-11-04:
Hello,  I was able to install PlatformIO IDE using Atom but at the moment of initializing atom I get these two errors
[image]

### **PlatformIO: Unable to create a virtualenv.**
Traceback (most recent call last):
  File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\Game\.atom\...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2
Close All
## **PlatformIO tool is not available.**
Can not find `platformio` command. Please install it using `pip install platformio` or enable built-in PlatformIO tool in `Menu: PlatformIO > Settings > platformio-ide` package.
Details:
***************************************************************
*** atom version 1.11.2 ******

************************************************************
*** this is my python information 
PS C:\Users\Game\mystuff> python -v
# installing zipimport hook
import zipimport # builtin
# installed zipimport hook
# C:\Python27\lib\site.pyc matches C:\Python27\lib\site.py
import site # precompiled from C:\Python27\lib\site.pyc
# C:\Python27\lib\os.pyc matches C:\Python27\lib\os.py
import os # precompiled from C:\Python27\lib\os.pyc
import errno # builtin
import nt # builtin
# C:\Python27\lib\ntpath.pyc matches C:\Python27\lib\ntpath.py
import ntpath # precompiled from C:\Python27\lib\ntpath.pyc
# C:\Python27\lib\stat.pyc matches C:\Python27\lib\stat.py
import stat # precompiled from C:\Python27\lib\stat.pyc
# C:\Python27\lib\genericpath.pyc matches C:\Python27\lib\genericpath.py
import genericpath # precompiled from C:\Python27\lib\genericpath.pyc
# C:\Python27\lib\warnings.pyc matches C:\Python27\lib\warnings.py
import warnings # precompiled from C:\Python27\lib\warnings.pyc
# C:\Python27\lib\linecache.pyc matches C:\Python27\lib\linecache.py
import linecache # precompiled from C:\Python27\lib\linecache.pyc
# C:\Python27\lib\types.pyc matches C:\Python27\lib\types.py
import types # precompiled from C:\Python27\lib\types.pyc
# C:\Python27\lib\UserDict.pyc matches C:\Python27\lib\UserDict.py
import UserDict # precompiled from C:\Python27\lib\UserDict.pyc
# C:\Python27\lib\_abcoll.pyc matches C:\Python27\lib\_abcoll.py
import _abcoll # precompiled from C:\Python27\lib\_abcoll.pyc
# C:\Python27\lib\abc.pyc matches C:\Python27\lib\abc.py
import abc # precompiled from C:\Python27\lib\abc.pyc
# C:\Python27\lib\_weakrefset.pyc matches C:\Python27\lib\_weakrefset.py
import _weakrefset # precompiled from C:\Python27\lib\_weakrefset.pyc
import _weakref # builtin
# C:\Python27\lib\copy_reg.pyc matches C:\Python27\lib\copy_reg.py
import copy_reg # precompiled from C:\Python27\lib\copy_reg.pyc
# C:\Python27\lib\traceback.pyc matches C:\Python27\lib\traceback.py
import traceback # precompiled from C:\Python27\lib\traceback.pyc
# C:\Python27\lib\sysconfig.pyc matches C:\Python27\lib\sysconfig.py
import sysconfig # precompiled from C:\Python27\lib\sysconfig.pyc
# C:\Python27\lib\re.pyc matches C:\Python27\lib\re.py

import re # precompiled from C:\Python27\lib\re.pyc
# C:\Python27\lib\sre_compile.pyc matches C:\Python27\lib\sre_compile.py
import sre_compile # precompiled from C:\Python27\lib\sre_compile.pyc
import _sre # builtin
# C:\Python27\lib\sre_parse.pyc matches C:\Python27\lib\sre_parse.py
import sre_parse # precompiled from C:\Python27\lib\sre_parse.pyc
# C:\Python27\lib\sre_constants.pyc matches C:\Python27\lib\sre_constants.py
import sre_constants # precompiled from C:\Python27\lib\sre_constants.pyc
import _locale # builtin
# zipimport: found 132 names in C:\Python27\lib\site-packages\setuptools-21.0.0-py2.7.egg
# C:\Python27\lib\locale.pyc matches C:\Python27\lib\locale.py
import locale # precompiled from C:\Python27\lib\locale.pyc
import encodings # directory C:\Python27\lib\encodings
import encodings # directory C:\Python27\lib\encodings
# C:\Python27\lib\encodings\__init__.pyc matches C:\Python27\lib\encodings\__init__.py
import encodings # precompiled from C:\Python27\lib\encodings\__init__.pyc

# C:\Python27\lib\codecs.pyc matches C:\Python27\lib\codecs.py
import codecs # precompiled from C:\Python27\lib\codecs.pyc
import _codecs # builtin
# C:\Python27\lib\encodings\aliases.pyc matches C:\Python27\lib\encodings\aliases.py
import encodings.aliases # precompiled from C:\Python27\lib\encodings\aliases.pyc
import operator # builtin
# C:\Python27\lib\functools.pyc matches C:\Python27\lib\functools.py
import functools # precompiled from C:\Python27\lib\functools.pyc
import _functools # builtin
# C:\Python27\lib\encodings\cp1252.pyc matches C:\Python27\lib\encodings\cp1252.py
import encodings.cp1252 # precompiled from C:\Python27\lib\encodings\cp1252.pyc
Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.

regards

ivankravets, 2016-11-04:
Try `pip install -U pip` and restart IDE and try again.

smunozeyquem, 2016-11-04:
Hi ivankravets, I did what you told me and a got this error:

PS C:\Users\Game\mystuff> pip install -U pip
Exception:
Traceback (most recent call last):

  File "C:\Python27\lib\site-packages\pip\basecommand.py", line 211, in main
    status = self.run(options, args)
  File "C:\Python27\lib\site-packages\pip\commands\install.py", line 294, in run
    requirement_set.prepare_files(finder)
  File "C:\Python27\lib\site-packages\pip\req\req_set.py", line 334, in prepare_files
    functools.partial(self._prepare_file, finder))
  File "C:\Python27\lib\site-packages\pip\req\req_set.py", line 321, in _walk_req_to_install
    more_reqs = handler(req_to_install)
  File "C:\Python27\lib\site-packages\pip\req\req_set.py", line 409, in _prepare_file
    req_to_install, finder)
  File "C:\Python27\lib\site-packages\pip\req\req_set.py", line 365, in _check_skip_installed
    finder.find_requirement(req_to_install, self.upgrade)
  File "C:\Python27\lib\site-packages\pip\index.py", line 486, in find_requirement
all_versions = self._find_all_versions(req.name)
  File "C:\Python27\lib\site-packages\pip\index.py", line 404, in _find_all_versions
    index_locations = self._get_index_urls_locations(project_name)
  File "C:\Python27\lib\site-packages\pip\index.py", line 378, in _get_index_urls_locations
    page = self._get_page(main_index_url)
  File "C:\Python27\lib\site-packages\pip\index.py", line 818, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "C:\Python27\lib\site-packages\pip\index.py", line 928, in get_page
    "Cache-Control": "max-age=600",
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\sessions.py", line 477, in get
    return self.request('GET', url, **kwargs)
  File "C:\Python27\lib\site-packages\pip\download.py", line 373, in request

    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\sessions.py", line 465, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\sessions.py", line 605, in send
    r.content
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\models.py", line 750, in content
    self._content = bytes().join(self.iter_content(CONTENT_CHUNK_SIZE)) or bytes()
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\models.py", line 673, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\packages\urllib3\response.py", line 307, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "C:\Python27\lib\site-packages\pip\_vendor\requests\packages\urllib3\response.py", line 243, in read
    data = self._fp.read(amt)
 File "C:\Python27\lib\site-packages\pip\_vendor\cachecontrol\filewrapper.py", line 54, in read
    self.__callback(self.__buf.getvalue())

  File "C:\Python27\lib\site-packages\pip\_vendor\cachecontrol\controller.py", line 244, in cache_response
    self.serializer.dumps(request, response, body=body),
  File "C:\Python27\lib\site-packages\pip\download.py", line 276, in set
    return super(SafeFileCache, self).set(*args, **kwargs)
  File "C:\Python27\lib\site-packages\pip\_vendor\cachecontrol\caches\file_cache.py", line 99, in set
    with self.lock_class(name) as lock:
  File "C:\Python27\lib\site-packages\pip\_vendor\lockfile\mkdirlockfile.py", line 18, in __init__
    LockBase.__init__(self, path, threaded, timeout)
  File "C:\Python27\lib\site-packages\pip\_vendor\lockfile\__init__.py", line 189, in __init__
    hash(self.path)))
  File "C:\Python27\lib\ntpath.py", line 85, in join
    result_path = result_path + p_path
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe1 in position 7: ordinal not in range(128)

Also when I open the terminal (Plataformio ide terminal), I get this error too:
[image]

ivankravets, 2016-11-04:
Something is wrong with your machine. Try
```
pip install --no-cache-dir -U pip
```

smunozeyquem, 2016-11-04:
PS C:\Users\Game\mystuff> pip install --no-cache-dir -U pip
Collecting pip
  Downloading pip-9.0.0-py2.py3-none-any.whl (1.3MB)
    100% |################################| 1.3MB 561kB/s
Installing collected packages: pip
  Found existing installation: pip 7.1.2
    Uninstalling pip-7.1.2:

      Successfully uninstalled pip-7.1.2
Exception:
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\pip\basecommand.py", line 211, in main
    )
  File "C:\Python27\lib\site-packages\pip\commands\install.py", line 311, in run
    requirement_set, args, options, finder, session, self.name,
  File "C:\Python27\lib\site-packages\pip\req\req_set.py", line 657, in install
    '--upgrade to upgrade): %s',
  File "C:\Python27\lib\site-packages\pip\req\req_install.py", line 729, in commit_uninstall
    options = {"delimiters": ('=', )}
  File "C:\Python27\lib\site-packages\pip\req\req_uninstall.py", line 152, in commit
    )
  File "C:\Python27\lib\site-packages\pip\_vendor\retrying.py", line 49, in wrapped_f
    return Retrying(*dargs, **dkw).call(f, *args, **kw)

  File "C:\Python27\lib\site-packages\pip\_vendor\retrying.py", line 212, in call
    raise attempt.get()
  File "C:\Python27\lib\site-packages\pip\_vendor\retrying.py", line 247, in get
    six.reraise(self.value[0], self.value[1], self.value[2])
  File "C:\Python27\lib\site-packages\pip\_vendor\retrying.py", line 200, in call
    attempt = Attempt(fn(*args, **kwargs), attempt_number, False)
  File "C:\Python27\lib\site-packages\pip\utils\__init__.py", line 90, in rmtree
    try:
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Python27\lib\shutil.py", line 252, in rmtree
onerror(os.remove, fullname, sys.exc_info())
  File "C:\Python27\lib\site-packages\pip\utils\__init__.py", line 102, in rmtree_errorhandler
    onerror=rmtree_errorhandler)
WindowsError: [Error 5] Access is denied: 'c:\\users\\game\\appdata\\local\\temp\\pip-sdxsqc-uninstall\\python27\\scripts\\pip.exe'

ivankravets, 2016-11-05:
Please try again the same command. Do you run it with Admin privilege? Run from generic user.

smunozeyquem, 2016-11-05:
I think I fix the problem, the path for python was wrong in my computer, so I change the path and everything install with no problem.

now the only error that I'm receiving is this one, at the moment that I open Atom:

PlatformIO: Unable to create a virtualenv.

Traceback (most recent call last):
File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
main()
File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 705, in main
symlink=options.symlink)
File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
download=download,
File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
call_subprocess(cmd, show_stdout=False, extra_env=env)
File "C:\Users\Game\AppData\Local\Temp\d-116104-10008-1fp9rjl.n18vbzkt9\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
% (cmd_desc, proc.returncode))
OSError: Command C:\Users\Game.atom...v\Scripts\python.exe -c "import sys, pip; sys...d"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

smunozeyquem, 2016-11-06:
so today, I install windows 10 in my pc, and follow the installation guide, and I got no problem with platformIO.
So now I'm testing the program with a nodemcu esp8266, and it works fine. The code is updating to the micro controller, Although every time that I update a code, in one line of the terminal appear an error, and I would like to know why is happing or if it's normal.
 here is part of the code that appear in the terminal:

[11/05/16 22:51:49] Processing nodemcuv2 (platform: espressif8266, board: nodemcuv2, framework: arduino)
 
[11/05/16 22:51:49] Processing nodemcuv2 (platform: espressif8266, board: nodemcuv2, framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Verbose mode can be enabled via `-v, --verbose` option
Collected 23 compatible libraries

Looking for dependencies...
Project does not have dependencies
Collected 23 compatible libraries
Looking for dependencies...
Project does not have dependencies
Looking for upload port...
Auto-detected: COM3
Uploading .pioenvs\nodemcuv2\firmware.bin
Looking for upload port...
Auto-detected: COM3
Uploading .pioenvs\nodemcuv2\firmware.bin
error: Failed to open COM3
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
 [ERROR] Took 1.84 seconds
Uploading 226352 bytes from .pioenvs\nodemcuv2\firmware.bin to flash at 0x00000000
................................................................................ [ 36% ]
................................................................................ [ 72% ]
 
"after 72%, it reached the 100% and the program end to upload"

So in resume the program work fines the codes that I tested until now works fine in te micro controller, even with this error that appear at the moment of updating the code.

thanks in for this amazing program and sorry for my english  xD!
regards

ivankravets, 2017-02-21:
Could you try new PlatformIO IDE 2.0 Installer?

Menu: PlatformIO > Home > Try PlatformIO IDE 2.0 Preview.

Thanks!

commit cecd5554 (2017-02-21): Use “conda” (if exists) to create virtual environment // Resolve #152 Resolve #174 Resolve #176 Resolve #209 Resolve #218
closed 2017-02-21 by ivankravets