platformio/platformio-atom-ide #179: Looping Rebuilt C/C++ index

hwacookie, 2016-11-12:
IDE 1.6.0 | CLI 3.1.0
All Atom packages are up-to-date (as of 11/12/2016)
The "Rebuilding library" is running multiple times, sometimes 30 minutes in a row, and it sometimes even seems to overtake itself.
I'm aware that the issue description is probably not very helpful, and I'd be happy to provide more information as needed.
Or is this maybe the wrong place to report this, anyway?

[image]

VilleP, 2016-11-24:
Have the same problem with OSX 10.9.5 and IDE 1.6.0 | CLI 3.2.0a7

ivankravets, 2016-11-24:
Could you share `platformio.ini`?

VilleP, 2016-11-24:
Sure:

[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps =
  /Volumes/MacintoshHD/Users/Documents/Arduino/libraries/RFM69
  /Volumes/MacintoshHD/Users/Documents/Arduino/libraries/SPIFlash
  /Volumes/MacintoshHD/Users/Documents/Arduino/libraries/WirelessHEX69
  /Volumes/MacintoshHD/Users/Documents/Arduino/libraries/SimpleTimer

I have also disabled "Automatically rebuild C/C++ Project Index" which end up in an endless loop of rebuilt messages and several python processes which consume 100% of CPU resources.

ivankravets, 2016-11-24:
> lib_deps =
> /Volumes/MacintoshHD/Users/Documents/Arduino/libraries/RFM69
> ...

What does it mean? Please read docs http://docs.platformio.org/en/stable/projectconf.html#projectconf-lib-deps

It should be as:

```ini
lib_extra_dirs = /Volumes/MacintoshHD/Users/Documents/Arduino/libraries/
```

instead of all lines

2. Please share here console output from IDE: View > Developer > Toggle Developer Tools`. You should see some log as "File watch added: ... Directory watch added"...

VilleP, 2016-11-24:
Thanks, changing to "lib_extra_dirs"  seems to solve the issue. Is a multi-line definition also possible?

ivankravets, 2016-11-24:
@hwacookie please share your `platformio.ini`

ivankravets, 2016-11-24:
@VilleP See http://docs.platformio.org/en/stable/projectconf.html#lib-extra-dirs

hwacookie, 2016-11-24:
Will do, I'm still at work, though :(

hwacookie, 2016-11-28:
Sorry for the late reply. I tried to upgrade my Ubuntu from 14.04 to 16.04 in the meantime :-/
Anyway, here it is:

I actually do have more than one platformio.ini files open within atom, i.e. one project called "SensorNet". with a platformio.ini looking like this:

[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
lib_deps=
  PrintEx
  VirtualWire
  Adafruit Unified Sensor
  DHT sensor library
  Adafruit DHT Unified

... and then another three projects with platformios looking like this:

[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
upload_port=/dev/ttyUSB1
lib_deps=
  file://../SensorNet
  PrintEx
  VirtualWire
  Adafruit Unified Sensor
  DHT sensor library
  Adafruit DHT Unified

So, what I'm trying to accomplish is to have one shared library in "SensorNet" (base classes, shared stuff) and the multiple other projects who use this base project. Is there a better way to accomplish this? I'd really like to have the base project kind of "open for editing" because it is far from stable yet.

ivankravets, 2016-11-28:
@hwacookie Please remove this line `file://../SensorNet`. You need http://docs.platformio.org/en/stable/projectconf.html#projectconf-lib-extra-dirs

hwacookie, 2016-11-28:
Wow, that was fast!
Unfortunatly, this isn't really working for me. I moved the "SensorNet" directory in a sub-folder called shared, and set the lib_extra_dirs to that "shared" folder. My projects platformio.ini now looks like...

[env:custom_lib_dirs]
platform = atmelavr
lib_extra_dirs = shared

[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
upload_port = /dev/ttyUSB1
lib_deps=
  PrintEx
  VirtualWire
  Adafruit Unified Sensor
  DHT sensor library
  Adafruit DHT Unified

Unfortunaly, it does not find anything from the "SensorNet" library anymore (the includes fail)

If I use the platformio.ini from above (with the "file://...") the imports are ok.
Right now, it also does not go into that rebuild loop. Anyway, why shouldn't I use the "file://" thing if it is already there? To me, that looks like the proper way of referencing another project?  Or is it meant for something else? Questions, questions, questions ...

BTW, thank you very much for all the effort you're putting into this!

ivankravets, 2016-11-28:
@hwacookie Pelase be careful and read http://docs.platformio.org/en/stable/projectconf.html#projectconf-lib-extra-dirs

You should place `lib_extra_dirs = shared` to `[env]` group.

P.S: I opened new feature request to allow using of `lib_extra_dirs` in global `[platformio]` group.

hwacookie, 2016-12-02:
I think I *did* read that. (See my post from four days ago) Unfortunatly it just does not work for me.
Is there any way to "include" the sources of one project to another? Like you can do in VisualStudio or Eclipse by referencing another project? 
Or can I maybe add a second "source" directory to a project? I'm totally fine if that results in the need to recompile all the referenced sources when you do a build. (actually, I thought that this would be what the "file://" thing was meant for. Apparently it isn't, but: why is it there then?

ivankravets, 2016-12-02:
@hwacookie Of course, you can do more than in Eclipse or VisualStudio. Please read http://docs.platformio.org/en/stable/projectconf.html#projectconf-lib-extra-dirs

```
[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
upload_port = /dev/ttyUSB1
lib_extra_dirs = dir1, dir2, dir2
lib_deps=
  PrintEx
  VirtualWire
  Adafruit Unified Sensor 
  DHT sensor library
  Adafruit DHT Unified
```

Please note that `dir1, dir 2, etc` ARE NOT LIBRARY DIRS. This is a directory that contains a list of libraries. For example, you have external libraries A & B that are located in directory C. The correct value should be `lib_extra_dirs = C`.

ivankravets, 2017-03-08:
Fixed in the latest PIO Core https://github.com/platformio/platformio-core/commit/58942c3f38ac2d48622784ebcc0b6fe37f7a8d0f

closed 2017-03-08 by ivankravets

joscha, 2017-04-11:
related to #356