platformio/platformio-atom-ide #180: Unable to execute: platformio

maru-sama, 2016-11-15:
Hello, 

After updating atom to the latest version (1.12.3) on macosx I suddenly have the issue that Building, Cleaning or Uploading fails with "Unable to execute: platformio"
Make sure cmd'platformio'.......
Looking at the path which is being searched shows, that the path to the built-in platformIO is missing. If If I then go to the platform-ide settings and toggle "Use builtin PlatformIO" (uncheck and check it again) everything works. It looks like that for some reason this is not setup correctly during startup.

ivankravets, 2016-11-16:
I can't reproduce this issue. Could you try to uninstall `platformio-ide` package from the Atom and install again? Please go to :gear: Settings > Packages

maru-sama, 2016-11-16:
No, re-installing does not fix the issue. I removed both ide and the terminal and reinstalled. But I still get the same error message.

ivankravets, 2016-11-16:
Is this option enabled " "Use builtin PlatformIO" (uncheck and check it again) "?

maru-sama, 2016-11-16:
Yes this is enabled, when I uncheck and check it again it works for this sessions. When I close atom and reopen it, this checkbox is STILL checked but I get the error. Doing an uncheck-check makes it work again.

maru-sama, 2016-11-16:
In the error message Binaries are found in these folders: ......
does not include the pioenvs directory.
For me it looks like this is not set correctly during startup but works later on when you toggle the checkbox.

maru-sama, 2016-11-16:
Ok, I tried to debug this.
maintenance.js correctly sets process.env.PATH and prepends the correct folder a debug output clearly shows this. BUT later on the build package errors out with above mentioned error and in its output process.env.PATH does NOT contain the folder.

maru-sama, 2016-11-16:
Ok I found the issue. Atom changed the way it starts up on the mac ( https://github.com/atom/atom/pull/13200). Adding "activationHooks": ["core:loaded-shell-environment"] to package.json as mentioned in the ticket fixes the issue for me.

ivankravets, 2016-11-17:
Thanks, added it to manifest.

closed 2016-11-17 by ivankravets

bobyk, 2020-09-01:
https://docs.platformio.org/en/latest/core/installation.html#method-2

commit cda10808 (2016-11-17): Add activationHooks with “core:loaded-shell-environment” // Resolve #180