platformio/platformio-atom-ide #188: How to disable PlatformIO in Atom?

urbanmichal, 2016-12-16:
How to disable PlatformIO in Atom?

ivankravets, 2017-02-21:
Settings ⚙️ > Packages > PlatformIO IDE > Disable

closed 2017-02-21 by ivankravets

tomkcook, 2017-08-04:
I have to say, it does feel a lot like a PlatformIO takes over your Atom installation.  I use Atom for a lot of things - PlatformIO is one of them, but it's an occasional one.  Now, when I'm hacking on ESP32 things on a weekend, it's great.  But during the week when I'm hacking other stuff that requires me to pop up Atom to edit some python every hour or so, it feels like way too often that PlatformIO gets in the way.  Disabling the "Show PlatformIO Home on startup", "Automatically Update PlatformIO IDE" and "Check PIO Account authentication status on IDE startup" settings does get rid of most of it, but these feel kind of like they either should be defaults or it should be possible to disable the automatic display of features from those features (ie the login form should have a 'Don't show me this again' checkbox).

ivankravets, 2017-08-05:
https://github.com/platformio/platformio-atom-ide/blob/develop/lib/config.js#L105

tenten8401, 2017-08-17:
I personally think there should be a hotkey or menu option to quickly disable/enable PlatformIO in the atom installation, or even just to hide the toolbar & whatnot immediately.

miroslavpetrov, 2018-03-23:
@tomkcook I feel the same way. I use Atom for many other things than Platformio and the left-side toolbar is quite irritating...So is there a solution to enable/disable platformio with shortcut or key combination?

cquiroz, 2019-03-29:
I agree, I use PlatformIO maybe 10% of my time but I need to see it always. A shortcut would be great

ivankravets, 2019-03-30:
Please sorry for the issue. We highly recommend using [PlatformIO IDE for VSCode](https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide). It has muuuuch better integration with PlatformIO and resolves this issue.