platformio/platformio-atom-ide #190: alloca.h missing in ststm32

alm494, 2016-12-29:
I have installed the last version of PlatformIO IDE on Windows7x64 , then add a new project, selected board type "genericSTM32F103C8", everything installed succesfully.
An empty main.cpp template compiles with the next output:

> Compiling .pioenvs\genericSTM32F103C8\FrameworkArduinoVariant\wirish\boards.o
> Compiling .pioenvs\genericSTM32F103C8\FrameworkArduinoVariant\wirish\boards_setup.o
> arm-none-eabi-g++: error: CreateProcess: No such file or directory
> *** [.pioenvs\genericSTM32F103C8\src\main.o] Error 1
> In file included from C:\users\nnb05524\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/WString.h:26:0,
> from C:\users\nnb05524\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Print.h:27,
> from C:\users\nnb05524\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/HardwareSerial.h:38,
> 
> from C:\users\nnb05524\.platformio\packages\framework-arduinoststm32\STM32F1\variants\generic_stm32f103c\board.cpp:40:
> c:\users\nnb05524\.platformio\packages\installing-woyf2c-package\arm-none-eabi\include\stdlib.h:21:20: fatal error: alloca.h: No such file or directory
> #include <alloca.h>
> ^
> compilation terminated.
> *** [.pioenvs\genericSTM32F103C8\FrameworkArduinoVariant\board.o] Error 1
> In file included from C:\users\nnb05524\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/WString.h:26:0,
> from C:\users\nnb05524\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/Print.h:27,
> from C:\users\nnb05524\.platformio\packages\framework-arduinoststm32\STM32F1\cores\maple/usb_serial.h:34,
> from C:\users\nnb05524\.platformio\packages\framework-arduinoststm32\STM32F1\variants\generic_stm32f103c\wirish\boards_setup.cpp:43:
> c:\users\nnb05524\.platformio\packages\installing-woyf2c-package\arm-none-eabi\include\stdlib.h:21:20: fatal error: alloca.h: No such file or directory
> #include <alloca.h>
> ^
> compilation terminated.
> *** [.pioenvs\genericSTM32F103C8\FrameworkArduinoVariant\wirish\boards_setup.o] Error 1
> arm-none-eabi-g++: error: CreateProcess: No such file or directory
> *** [.pioenvs\genericSTM32F103C8\FrameworkArduinoVariant\wirish\boards.o] Error 1
>  [ERROR] Took 2.50 seconds
>

ivankravets, 2016-12-29:
Something is wrong with your installation. Please remove this folder and restart IDE

* `c:\users\nnb05524\.platformio`

closed 2016-12-29 by ivankravets

alm494, 2016-12-29:
thank you.
you was right:

> PlatformManager: Installing ststm32
> Downloading...
> Unpacking...
> ststm32 @ 1.6.0 has been successfully installed!
> The platform 'ststm32' has been successfully installed!
> 
> The rest of packages will be installed automatically depending on your build environment.
> PackageManager: Installing framework-arduinoststm32 @ ~2.0.0
> Downloading...
> Unpacking...
> framework-arduinoststm32 @ 2.0.0 has been successfully installed!
> PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.40803.0,<1.40805.0
> Downloading...
> Unpacking...
> Warning! Package Mirror: [Error 5] 
> Looking for other mirror...
> Downloading...
> Unpacking...
> Warning! Package Mirror: [Error 5] 
> Looking for other mirror...
> Downloading...
> Unpacking...
> toolchain-gccarmnoneeabi @ 1.40804.0 has been successfully installed!