platformio/platformio-atom-ide #192: Serial monitor not working (windows)

rjstone, 2016-12-31:
I have a simple serial command line parser installed on my board.

Using the PlatformIO serial monitor I connect successfully but no matter what I type I get:

```
Line read:
Unknown command.
```

It is as though the only character getting sent is the newline. This happens no matter what serial settings I use. Raw mode makes no difference. Parity, etc makes no difference.

If I use the Arduino IDE serial monitor I get this: (example: typing 'help' then enter)

```
Line read: help
Available commands:
   help - show this help
```

This seems to be a problem with the `pio` command because the same thing happens even if I run it in a regular shell (outside of atom).

ivankravets, 2017-01-05:
Could you provide simple sketch to debug this issue?

rjstone, 2017-01-18:
Sorry for the delay.

I suspect this is some sort of bizarre issue on my end. A sketch suitable for debugging the problem would be the example code for the contributed SerialCmd Arduino library.

I actually installed the PlatformIO terminal just to use as a shell console long before I ever installed the rest of PlatformIO. I ran into some sort of problem with not having pyserial installed or some other Python dependency installed.

I'm just going to close this until I can uninstall Python, uninstall the terminal package, and then try reinstalling the PlatformIO terminal without any of the dependencies it wanted to see what it tells me to do.

closed 2017-01-18 by rjstone