platformio/platformio-atom-ide #195: PlatformIO IDE insists I want minimap

micr0me, 2017-01-07:
Minimap is a package I find little use for (and find rather annoying). I've tried disabling minimap under Settings->Packages -- that doesn't work. I've tried uninstalling the package. However, every time I start atom PlatformIO installs it again.

Can someone please point me to a way of disabling this? I apologize if it's obvious.

ivankravets, 2017-01-07:
This is a bug, thank you! 

Temporary solution is to manually remove/comment this line

* https://github.com/platformio/platformio-atom-ide/blob/develop/lib/config.js#L103

This file is located in `~/.atom/packages/platformio-ide/lib/config.js`.

micr0me, 2017-01-07:
That works great. Thank you!

ivankravets, 2017-02-21:
Fixed in a new PlatformIO IDE 2.0 installer

closed 2017-02-21 by ivankravets

surdu, 2017-06-15:
This is still an issue in v2.0.0-beta.6

ivankravets, 2017-06-15:
Just disable Minimap package

surdu, 2017-06-16:
Isn't desireable to have an option to disable minimap requirement ?

I could try implementing it ...