platformio/platformio-atom-ide #199: Object.extname is deprecated.

blakehurlburt, 2017-01-13:
Argument to `path.extname` must be a string
```
Object.extname (/usr/share/atom/resources/app.asar/src/electron-shims.js:20:10)
<unknown> ($HOME/.atom/packages/platformio-ide/lib/main.js:274:39)
Workspace.observeTextEditors (/usr/share/atom/resources/app.asar/src/workspace.js:329:9)
Object.setupCommands ($HOME/.atom/packages/platformio-ide/lib/main.js:272:43)
<unknown> ($HOME/.atom/packages/platformio-ide/lib/main.js:89:20)
```

ivankravets, 2017-01-13:
What is a version of PIO IDE? How to reproduce this issue?

ivankravets, 2017-01-14:
@sobodash what is your PIO IDE version?

ivankravets, 2017-01-14:
This is a stupid warning. Please ignore it. Do you see it after Atom restart?

sobodash, 2017-01-15:
I see it every time I open Atom in the lower right corner. I am on IDE 1.7.1 and Core 3.2.1

lukehoggett, 2017-01-16:
I get this warning after upgrading to Atom 1.13.0 in the deprecations screen

chedongzhe, 2017-01-23:
Argument to path.extname must be a string
Report Issue
Object.extname - C:\Users\CheDongZhe\AppData\Local\atom\app-1.13.0\resources\app.asar\src\electron-shims.js:20:10
<unknown> - C:\Users\CheDongZhe\.atom\packages\platformio-ide\lib\main.js:274:39
Workspace.observeTextEditors - C:\Users\CheDongZhe\AppData\Local\atom\app-1.13.0\resources\app.asar\src\workspace.js:329:9
Object.setupCommands - C:\Users\CheDongZhe\.atom\packages\platformio-ide\lib\main.js:272:43
<unknown> - C:\Users\CheDongZhe\.atom\packages\platformio-ide\lib\main.js:89:20

closed 2017-01-24 by ivankravets

shanevanj, 2017-02-01:
Guys - the error is still there - can it be patched out?

ivankravets, 2017-02-01:
@shanevanj soon! :) Stay tuned with us on https://twitter.com/PlatformIO_Org

commit 68b37464 (2017-01-24): Skip editors with undefined paths

Resolves #199
Resolves #202
Resolves #203
Resolves #205